nanopb/nanopb #159: proto3 support plans?

Sunib, 2015-06-19:
Let me start with a big thank you for everyone that contributes to this project, it's a wonderfull library that works like a charm.

Now the issue: Google is introducing a new version of protocol buffer language: [Proto3](https://developers.google.com/protocol-buffers/docs/proto3). As far as I can see it's an update to the proto language itself. The binary encoding would not have to change (much) I presume.

Would this be something that is going to be included in nanopb? Would it have big implications?

PetteriAimonen, 2015-06-20:
https://groups.google.com/forum/#!topic/nanopb/1di-Ua50N3Q   Related discussion

ekarlso, 2015-11-05:
Any clue on this? I'm wanting to use protobuf 3..

PetteriAimonen, 2015-11-05:
Once issue #182 has been fixed, I think protobuf 3 should be usable for many purposes with nanopb. But a lot remains to be done before one can claim full support.

kamathba, 2015-11-12:
i was thinking of writing the pb_json module for encoding from/to json. Encoding seems relatively straightforward in my mind, but decoding requires writing a json parser. Any objection to snapshotting something like jsmn? https://github.com/zserge/jsmn MIT license. Could either modify it to decode while parsing instead of tokenizing, or just use the tokenized outputs similar to the simple example.

PetteriAimonen, 2015-11-12:
Jsmn seems to be quite small, and indeed copying it could help for 
customizing it to be more suitable for use in nanopb. Though using
it without modification is fine by me also, do whatever seems most
reasonable :)

License-wise I see no problem, atleast as long as jsmn is confined
to a module that can be selectively included. Maybe it should
go directly into pb_json.c, in order to keep the total numbers of
files small which IMO makes the structure of nanopb easier to 
grasp.

kamathba, 2015-11-12:
so we could have:

a) pb_json.c/h which defines the encode/decode for json.

b) we could also have pb_json_encode.c/h and pb_json_decode.c/h with the parsing stuff being included/referenced from the decode module.

c) Or we could build the json encode/decode into pb_encode.c/h and pb_decode.c/h with preprocessor macros.

I like c but it might make those files pretty large. b introduces a bunch of new files. Maybe option a is best :)

PetteriAimonen, 2015-11-13:
I like a) or b), maybe a) more.

kamathba, 2015-11-17:
What does everyone think about gRPC and supporting services?

winksaville, 2015-11-18:
As long as its causes no additional code foot print for those not using it,
I'd say its fine.

kamathba, 2015-11-24:
Been thinking about this for a little bit before starting too far on it. One interesting thing I think we might need specifically for enums is a lookup table of like struct foo { enum_t value; char* name; } enum_t_string_table [] = { { ENUM_VAL_1, "EnumVal1"}, {...}, ... }; or something like that.

I can't think of a great way of encoding aliases as anything but the first/default one if the enum allows aliases. I think regular field names can be stored in the field entries without compiling into extra space for people not using them  (have some of that done).

Meanwhile, for decoding, if we use something like jsmn and want to decode into jsmntok_t's, we'd need some sort of pb_json_istream_from_buffer(s) type of function that takes a buffer for the stream and another chunk of memory provided for the array of tokens. The number of tokens could be deterministic if it is a deterministic size message (no callback_t fields). The tokens could come from some fixed pool but I like that less since it forces a threading model or would require a porting layer. Haven't played with it enough to know what the resource/performance trade-offs are of decoding the stream from json into the structs on the fly.

PetteriAimonen, 2015-11-24:
It's starting to seem that this work should go into some dev-branch
which will eventually become nanopb-0.4.0. There will be several
breaking changes necessary along the way.

> One interesting thing I think we might need specifically for enums 
> is a lookup table

I agree. Doing #114 will free up the pointer field in the enum field
descriptors, this can then point to the enum lookup table.

> I can't think of a great way of encoding aliases as anything but the 
> first/default one if the enum allows aliases.

Makes sense.

> The number of tokens could be deterministic if it is a deterministic 
> size message (no callback_t fields).

Could nanopb call jsm_parse() several time to obtain a set of tokens? 
This way there would not need to be a buffer for all the tokens.

PetteriAimonen, 2017-03-05:
I'm closing this off, basic proto3 support has been included since nanopb-0.3.7.

Support for Any fields, json etc. is still missing, but should be filed as separate feature requests once someone needs them.

closed 2017-03-05 by PetteriAimonen