nanopb/nanopb #160: Variable-length payload with protobuf and nanopb

ncolomer, 2015-07-01:
I'm trying to send a serialized message through RF24 (nRF24L01+ module) and I need to know the size of serialized payload. Here is the code I use:

```cpp
pb_ostream_t stream = pb_ostream_from_buffer(buffer, sizeof(buffer));
bool status = pb_encode(&stream, MoteData_fields, &message);
```

Do you now if protobuf message length are variable? If so, how can we retrieve this dynamic size?
I'd expected the following would be enought:

```cpp
uint8_t size = stream.bytes_written;
```

But it seems to always return the max size as defined in the generated header file `MyMessage.pb.h`: 

```cpp
/* Maximum encoded size of messages (where known) */
#define MyMessage_size                            36
```

Any suggestion?

PetteriAimonen, 2015-07-01:
Hi,

> Do you now if protobuf message length are variable? If so, how can we 
> retrieve this dynamic size?

It is variable and stream.bytes_written should give the length.

> But it seems to always return the max size as defined in the generated 
> header file `MyMessage.pb.h`:
> Any suggestion?

Hmm, does the encode finish successfully? Maybe for some reason it is 
using more space and fails when the buffer gets full. This would be a 
bug in nanopb's max_size calculation code.

Otherwise maybe the message is constant and just happens to use up the 
maximum length?

--
Petteri

ncolomer, 2015-07-01:
Thanks for your answer Petteri

I'll do further investigations but the decoding in the other side returns true (and I correctly get my data)

Moreover, is there a way to know/deduct the cost of each field (from its pb type for instance) in the total size?

PetteriAimonen, 2015-07-01:
> Moreover, is there a way to know/deduct the cost of each field (from 
> its pb type for instance) in the total size?

Only manually. If you have a good use case for this, please tell.

PetteriAimonen, 2015-08-08:
Any further information on this? Otherwise I'll close it off.

ncolomer, 2015-08-09:
Sorry for the delay, I had no time to investigate further I'll let you know once I do.
I close the issue in the meantime.
Thanks!

closed 2015-08-09 by ncolomer