nanopb/nanopb #163: Support Any message type

vahid-sohrabloo, 2015-08-09:
Is nanopb support Any message type?
https://developers.google.com/protocol-buffers/docs/proto3?hl=en#any

PetteriAimonen, 2015-08-09:
No, there is no protobuf 3 support in nanopb so far.

PetteriAimonen, 2015-10-24:
Hmm, looking at the google/protobuf/any.proto file, looks like it might work even without any special support. Like this:

    message MyMsg {
         Any details = 1 [(nanopb).max_size = 5];
    }

and:

    MyMsg msg;
    pb_decode(...., &msg);
    if (strcmp(msg.details.type_url, "http://my-id-url.com/foo.proto") == 0)
    {
           Foo foobar;
           pb_istream_t s = pb_istream_from_buffer(msg.details.value.bytes, msg.details.value.size);
           pb_decode(&s, Foo_fields, &foobar);
    }

Jingzhe88, 2017-05-02:
Hi @PetteriAimonen, 

Could you point to an example to use any in the nanopb? I have searched around, but could not find any but this related issue. 

Thanks, 
James

wuzhenda, 2019-12-25:
need encode  example too

wuzhenda, 2019-12-30:
> Hmm, looking at the google/protobuf/any.proto file, looks like it might work even without any special support. Like this:
>
> ```
> message MyMsg {
>      Any details = 1 [(nanopb).max_size = 5];
> }
> ``` [...]

msg.details.type_url is a pb_callback_t type not string

PetteriAimonen, 2020-02-01:
I've now added a test case / example on Any type usage.

closed 2020-02-01 by PetteriAimonen