nanopb/nanopb #164: Issue with implementation defined enum size and signedness

hebnern, 2015-08-19:
A colleague recently ran into an interesting issue related to enum sizing and signedness. We are sending messages between two platforms which use different compilers. On platform A one of the enums generated by nanopb (with values between 0x00 and 0xFF) is treated as a 1 byte unsigned value, on platform B it is treated as a 4 byte value with unknown signedness. This is completely valid per the C spec. One of the enum values was 0x86 and causes an "integer too large" error inside of pb_dec_varint() since the top bit is set, and field->data_size == 1. We have temporarily worked around this by treating enums as uvarints instead, but have realized that this may present errors when sending signed values from a platform with smaller sized enums to one with larger sized ones.

A possible solution would be to have the generator add members to the enum with values of INT_MAX and INT_MIN to force them to always be treated as a signed int.

PetteriAimonen, 2015-08-19:
I don't quite like the idea of adding extra enum entries for every enum 
type.

Negative enum values are inefficiently coded by protobuf in any case,
so probably not very commonly used.

How would this sound to you:
1) Encode enums as signed 32 bits, except if the data_size is less than 
   4, then threat them as unsigned.
2) Generate static asserts from the generator if an enum field has 
   negative values defined. The static assert would check the sizeof
   of the field, and if less than 4 guide the user to add a dummy enum
   value of e.g. INT32_MAX to force the enum to 4 bytes length.

mschnaider, 2015-08-27:
I ran into this issue a little while ago, and have been trying out a fix internally that I can offer.  I added a compile time test for the storage size for enum and select VARINT or UVARINT as appropriate.  

I'm new to github, so please let me know if/how I should provide a patch that implements the following.  

My notes on my approach say:

"Update the definitions of PB_FIELD and PB_FIELD2 to take an additional
argument of the C typename for a given field.  

Update the PB_LTYPE_MAP_ family of macros to take the ctype as a argument.
For the ENUM variant, select at compile time between PB_LTYPE_VARINT and
PB_LTYPE_UVARINT based upon a signedness test of the underlying C enum
definition (as it will be stored by this compiler).

For packed enums, the compiler may use unsigned storage.  If the storage is
actually unsigned, then -1 cast to the enum type will not equal -1.  I.e.:

/* Compiler may use signed or unsigned storage for enums. */
#define PB_LTYPE_MAP_ENUM(ctype) \
    ((PB_LTYPE_VARINT) * ((ctype) -1 == -1) + \
     (PB_LTYPE_UVARINT) * ((ctype) -1 != -1))

In nanopb_generator.py, add the python ctype type name to the PB_FIELD2
macro population function.

PetteriAimonen, 2015-08-28:
Hmm, that is a promising option also. I just fear a bit how much various 
compilers will complain about the enum casts. They would be quite within 
their rights to give a warning about casting -1 to unsigned enum.

PetteriAimonen, 2015-09-12:
What about this solution: simply check in generator if the enum has negative values. If not, assign it to type UENUM and decode as unsigned. This is implemented in newest git master.

Currently only works for enums defined in the same file, but resolving #165 will allow it to work in multifile cases also.

PetteriAimonen, 2015-10-05:
Fix released in nanopb-0.3.4.

closed 2015-10-05 by PetteriAimonen

rolanddixon, 2017-06-14:
It seems that this is still an issue in the latest release version (0.3.8). On an Arm Cortex-M4 MCU, enums are encoded and decoded as 8-bit types. Trying to decode a stream coming from the MCU on a Windows machine (both with Visual Studio and MinGW) results in a 32-bit enum type. This breaks the whole decode as the offsets are now wrong and there is no indication from the decode routine that anything failed.

This happens regardless of the enum being imported into the .proto file or not...

PetteriAimonen, 2017-06-14:
@rolanddixon That sounds like a separate issue.. but as long as you compile the .pb.c with the same compiler as the rest of your code, the offsets should be automatically correct because they are taken with sizeof().

rolanddixon, 2017-06-14:
Oh ok. When I encode an enum on the MCU with gcc it is encoded as one byte. I can decode that same stream on the same MCU with no problems. When I try to decode the same stream from the MCU but on a Windows machine the enums are decoded as 4 bytes. An array of bytes follows the enum in the stream and I can see 3 of these array bytes in the little endian representation of the 4 byte enum along with 1 byte enum itself. The the array no longer contains these bytes which the enum has taken ownership of. The decoding of the (fixed_size) array must now surely be wrong as it would overlap onto the next section of memory.
This image shows the enum (deviceID surrounded in green) as 3 during the encode. This is all encoded into a oneof called ext. The array (top three elements surrounded in blue) .
[image]

This image shows the enum (deviceID surrounded in green) as 2114002179 (bytes are shown as well) This is a four byte value and contains the first three bytes of the array from the previous picture. The array now starts with the last three bytes from the picture before (which makes sense as they are now in the enum).
[image]

Hope this highlights the problem I am having.

rolanddixon, 2017-06-14:
At the moment I am encoding the enum as a fixed32 and I do a cast at runtime from the enum to a uint32_t then encode and vice versa for decode. This works fine.

PetteriAimonen, 2017-06-14:
@rolanddixon If you can get a reproducible example I can compile, please post it as its own issue report or on the forum. Based on that image its quite hard to tell what is going on, but it doesn't seem to match the issue in this bug.

rolanddixon, 2017-06-14:
Ahh, I thought my pictures were great... Ok, will open a new issue and send some example code.