nanopb/nanopb #165: max encoded size doesn't appear when items in a oneof are imported

brathke, 2015-08-20:
This problem can be demonstrated using the tests/oneof project. If I compile it "as is" I will see max size information in oneof.pb.h as follows:
```
/* Maximum encoded size of messages (where known) */
#define SubMessage_size                          88
#define OneOfMessage_size                        113
#define PlainOneOfMessage_size                   90
```

However, if I move SubMessage to a new file, SubMessage.proto:
```
import 'nanopb.proto';
message SubMessage
{
    repeated int32 array = 1 [(nanopb).max_count = 8];
}
```

And then import it into oneof.proto:
```
import 'nanopb.proto';
import 'SubMessage.proto';

//message SubMessage
//{
//    repeated int32 array = 1 [(nanopb).max_count = 8];
//}

/* Oneof in a message with other fields */
message OneOfMessage
{
    required int32 prefix = 1;
    oneof values
    {
        int32 first = 5;
        string second = 6 [(nanopb).max_size = 8];
        SubMessage third = 7;
    }
    required int32 suffix = 99;
}

/* Oneof in a message by itself */
message PlainOneOfMessage
{
    oneof values
    {
        int32 first = 5;
        string second = 6 [(nanopb).max_size = 8];
        SubMessage third = 7;
    }
}
```

After compiling, the new oneof.pb.h does not have any entries for max encoded sizes. i.e. a comment line without any definitions:
```
/* Maximum encoded size of messages (where known) */

```

Is this expected behavior?

PetteriAimonen, 2015-08-21:
Yes, currently this is expected behaviour. The maximum size has to be 
computed as a maximum of the possible oneofs, but when several files are 
involved the sizes are not known at generator.

Possible fixes:
1) Use some C preprocessor magic to calculate the maximum at runtime.
2) Make generator aware of multiple files at once (this could also help 
   with some other issues).

Either of these is currently a fair bit of work compared to the benefit 
so probably not going to be implemented very soon.

brathke, 2015-08-21:
Thanks for your response. I notice that imported items outside of the oneof show up like this
```
 #define OneOfMessage_size                        (119 + SubMessage2_size)
```

In fix #1, you mention "at runtime", can't this sort of thing be done at compile time by adding a max macro and then doing something like the following?
```
 #define OneOfMessage_size                        (119 + MAX(SubMessage2_size,MAX(...etc)
```

If not, is there another way I can determine the max size of this type of message? Thanks

PetteriAimonen, 2015-08-21:
Yes, sorry, in 1) I meant to write "at compile time".

PetteriAimonen, 2015-09-12:
Seems that adding multifile support to generator should be quite easy when it is run in the plugin mode.

Previously with the two-step generation (first protoc to foo.pb, then nanopb_generator.py) it was not easy to get the data for the included files. This is still true, but I guess solving it for the single-step protoc plugin case is worth it.

PetteriAimonen, 2015-10-05:
Fix released in nanopb-0.3.4.

closed 2015-10-05 by PetteriAimonen