nanopb/nanopb #167: Add syntax specification to all .proto files

PetteriAimonen, 2015-09-13:
Newest protoc complains if the .proto file does not have a syntax specification:

[libprotobuf WARNING google/protobuf/compiler/parser.cc:492] No syntax specified for the proto file. Please use 'syntax = "proto2";' or 'syntax = "proto3";' to specify a syntax version. (Defaulted to proto2 syntax.)

Should add this line to all files in nanopb.

PetteriAimonen, 2015-10-05:
Fix released in nanopb-0.3.4.

closed 2015-10-05 by PetteriAimonen

denravonska, 2016-10-07:
At bc2c47e I get:

`         ********************************************************************
         *** Failed to import the protocol definitions for generator.     ***
         *** You have to run 'make' in the nanopb/generator/proto folder. ***
         ********************************************************************

Traceback (most recent call last):
  File "/Users/marco/dev/exqbe/Bridge/nanopb/../../Vendor/nanopb/generator/nanopb_generator.py", line 33, in <module>
    import proto.nanopb_pb2 as nanopb_pb2
  File "/Users/marco/dev/exqbe/Vendor/nanopb/generator/proto/nanopb_pb2.py", line 26, in <module>
    dependencies=[google_dot_protobuf_dot_descriptor__pb2.DESCRIPTOR,])
TypeError: __init__() got an unexpected keyword argument 'syntax'`

This is with protobuf 2.6.1 and 3.1.0. I have tried to clean and re-done make in 'generator' after switching versions.

PetteriAimonen, 2016-10-07:
@denravonska Sounds like your protoc version and python protobuf version are mismatched.
You can try to check that it is imported from the place where you expect:
python -c 'import google.protobuf; print(google.protobuf.__file__)'

If you have both protobuf 2.6.1 and 3.1.0 installed on your system, you have to be especially careful to keep the python path matched to the protoc version you use.

denravonska, 2016-10-07:
Aaaah, I had 2.6 installed via pip as well. It took precedence over the one from Homebrew.