nanopb/nanopb #168: encode and decode callbacks are not called independently

gtorres88, 2015-09-15:
If the callback system is used for a field, and both the encode and decode callbacks are populated, only the decode callback is called both when doing encode and decode.

If only one is populated, it will be called both when encoding and decoding.

They should be called independently of each other during the correct operation.

PetteriAimonen, 2015-09-15:
This is as designed. They are stored in a C union to conserve RAM, so only one can be set at a time.
Best way is to set the callbacks right before call to pb_decode or to pb_encode.

In my opinion, the small inconvenience is worth the saving of RAM (8 bytes per callback field). Opinions welcome, of course.

gtorres88, 2015-09-15:
I don't know how I missed that. Got it!

closed 2015-09-15 by gtorres88