nanopb/nanopb #172: nanopb 0.3.4 breaks message size calculation working in 0.3.3

gonzzor, 2015-10-05:
Hi,

The release of nanopb-0.3.4 seems to have broken previously working proto files. The commit 0b29baf5deaa4213c08ee71fa55d3d0b2ed709e4 solving issue #165 is causing the size calculation to fail.

I have a very simple example.
test.proto
```
import "submessage.proto";

message testmessage {
 optional submessage sub = 1;
}
```

submessage/submessage.proto
```
message submessage {
    required bytes data = 1;
}
```

submessage/submessage.options
```
submessage.data max_size: 16
```

Compiling this with protoc 2.5.0
protoc -I. -Isubmessage --plugin=protoc-gen-nanopb=nanopb/generator/protoc-gen-nanopb --nanopb_out=. test.proto

The test.pb.h is missing the define, testmessage_size.
Using a commit of nanopb before the faulty one produces this line in test.pb.h
```
#define testmessage_size                         (6 + submessage_size)
```

PetteriAimonen, 2015-10-05:
The immediate regression is now fixed.

Issue is that while nanopb_generator now finds the other .proto file, it does not find the associated .options file. The fix now uses the old expression generation when the submessage size is not known.

However there remains a hairy corner case if options are defined both in .proto and .options. In this case the encoded size can be known, but wrong, when the .options is not found. Properly fixing this will require upstream https://github.com/google/protobuf/pull/811

Another fix with nanopb-0.3.4 is to give -Isubmessage to nanopb:
`protoc -I. -Isubmessage --plugin=protoc-gen-nanopb=nanopb/generator/protoc-gen-nanopb --nanopb_out=-Isubmessage:. test.proto`

gonzzor, 2015-10-05:
Yes. I can confirm that the regression I saw is gone now.

PetteriAimonen, 2019-12-20:
I'll close this old issue. There have been several small fixes to finding .options files over the years, but the underlying complexity still exists.

Now in 0.4.0, `nanopb_generator.py` accepts `.proto` files directly which makes it easier to find the related `.options` file.

closed 2019-12-20 by PetteriAimonen