nanopb/nanopb #173: Protobuf structure shallow copy

mottigondabi, 2015-10-07:
Failed to find deep copy of a given protobuf structure.

For static fields, a simple memcpy() will indeed suffice.

But with dynamic fields it is more involved than that. Currently the
simplest way is to encode the message and decode it again, but obviously
that is not very efficient.

One can also use the functions from pb_common.h to go through the field
descriptors to implement the deep copy. But this has quite a many field
types to handle.

--Motti

baryluk, 2020-12-17:
I wish there was a deep copy available, as well merging (i.e. message `a` into `b`, with fields in `b` that already already set but overwritten by values from `a`, to be freed first (`free` or `pb_release + free` depending on type of field), and then copied into `a`. If `a` doesn't have specific field set, let the existing value in `b` for that field be preserved).

baryluk, 2021-01-16:
FYI. I implemented a deep copy (including merging into existing message) using descriptors. it ended pretty complex, about 400 lines of code (including debugging).

I believe I handled almost everything, and valgrind is also happy, but some features like repeated bytes fields, extensions and oneof are not supported. Also non-pointer fixed-max-width strings are untested. And there is some zero-sized allocations that need fixing.

If anybody is interested, I am attaching a work-in-progress version in the attachment (zlib license, just like nanopb).

I am dropping it here, just for some comments and suggestions.

```c
/*
 * Copyright (c) 2021 Witold Baryluk <witold.baryluk@gmail.com>
 *
 * This software is provided 'as-is', without any express or 
 * implied warranty. In no event will the authors be held liable 
 * for any damages arising from the use of this software.
 *
 * Permission is granted to anyone to use this software for any 
 * purpose, including commercial applications, and to alter it and 
 * redistribute it freely, subject to the following restrictions:
 *
 * 1. The origin of this software must not be misrepresented; you 
 *    must not claim that you wrote the original software. If you use 
 *    this software in a product, an acknowledgment in the product 
 *    documentation would be appreciated but is not required.
 *
 * 2. Altered source versions must be plainly marked as such, and 
 *    must not be misrepresented as being the original software.
 *
 * 3. This notice may not be removed or altered from any source 
 *    distribution.
 */

#pragma once
#ifndef _MANGO_SERVER_PB_COPY_H_
#define _MANGO_SERVER_PB_COPY_H_

#ifndef PB_ENABLE_MALLOC
#define PB_ENABLE_MALLOC
#endif
#include <pb.h>
#include <pb_common.h>  /* for pb_field_iter_{begin,next} */

#ifdef __cplusplus
extern "C" {
#endif

/* Deep copy message 'from' to 'to' recursively.
 *
 * If 'to' already has fields set, and 'merge' is 1, then
 * only if 'from' has a corresponding value, it is overwritten
 * (and if needed deallocated before hand). Otherwise,
 * it is cleared (and if needed deallocated / released).
 *
 * This function tries to minimize new allocations and reallocations,
 * by reusing some existing buffer in 'to', if possible.
 *
 * Nested merging works too, including repeated fields.
 * For example, it to.r[2].x is set, but from.r[2].x is not,
 * the to.r[2].x will be left as is.
 *
 * All pointers must be not NULL.
 *
 * 'from' and 'to' must be same message type.
 *
 * 'from' and 'to' must not share memory, be the same message,
 * or be in child-parent relationship (transitively) between each other.
 *
 * After deep_copy_generic finishes, all data in 'to' is indenepdent
 * from 'from'. They can be modified or released / freed in any order.
 *
[... 300 lines cut ...]

    if (!from) {
        return; /* Ignore NULL pointers. */
    }

    if (!pb_field_iter_begin(&iter_from, fields, (void*)from)) {  // Drop const from 'from'.
        return; /* Empty message type */
    }
    if (!pb_field_iter_begin(&iter_to, fields, to)) {
        return; /* Empty message type */
    }

    do {
        deep_copy_single_field(&iter_from, &iter_to, merge, indent);
        pb_field_iter_next(&iter_from);
    } while (pb_field_iter_next(&iter_to));
}

void pb_deep_copy_generic(const pb_msgdesc_t *const fields, const void *const from, void *const to, int merge) {
    assert(fields != NULL);
    assert(from != NULL);
    assert(to != NULL);
    deep_copy_generic_internal(fields, from, to, merge, /*indent=*/0);
}
```

PetteriAimonen, 2021-01-16:
@baryluk Thanks for sharing!

Indeed doing it using the currently available descriptor functions is pretty complex and duplicates a lot of the internal code for handling repeated structures.

I've been thinking for a while now that for the reallocated arrays, nanopb should probably have some kind of helper functions. Especially as reallocing for each addition is terribly slow on some platforms (#571).

baryluk, 2021-01-17:
I extended the code to cover more cases, including handling merge & non-merge intent, as well some support for optional, at least on the simple values. I still need to figure out the repeated bytes, optional sub-messages (with has_ field), and extensions. But these are not too hard.

I am pretty happy with the code, it is complex, but once it is written fully it can be refactored a bit to be simpler. The hardest part is dealing with all possibilities, optional, non-optional, repeated, has_, _count, bytes, strings, fixed-limit-arrays, merging, and making sure there are no memory leaks ;D

It took me quite a bit to get a good mental model of what various pieces in the descriptor and iterator really mean.

I think the next step it so make a comprehensive and extensive test suite for this functionality. I can imagine ~100 test cases, many with custom setup and verification logic.

baryluk, 2021-01-18:
After writing some more tests, I identified a number of bugs and issues in the current code posted above. But I got local copies that do fix most of it, and actually simplify the code.

Right now I am writing comprehensive tests (66 different type+attribute+option combinations that are permitted, i.e. FT_STATIC repeated with no max_count is not permitted, or string are not permitted in FT_STATIC without max_length, and few other weird cases), multiplied by various condition for merge/copy, source and destination state (empty, non-empty, smaller, same or bigger that the other for repeated fields), this expands to 840 distinct tests (for basic types I only test `uint32` and `uint32` as IS_8 (`uint8_t`). So not testing all the other sub-types, like sint64, sfixed64, float, double, enum, as they will exercise same code paths, and adding them would balloon cases to about 10000), grouped into 140 test functions. So far I implemented 30 test functions today (exercising 180 different code paths), but the rest is just copy and paste and adjusting checks (including equality of result, distinct pointers, no leaks, no corruption of other state, etc). Macros are helping quite a bit with that.

PetteriAimonen, 2021-01-18:
Conceivably it could be also combined with the fuzz test.

I'm not sure how this would be best to merge with nanopb core - it seems quite large to include by default. Maybe it is time to add some `pb_extras.c` module.

baryluk, 2021-01-18:
@PetteriAimonen Yeah, I noticed the fuzz test in nanopb, but I didn't take a look into it yet. It would be nice to also hook it up, but more for a sanity, and as a combination with clang AddressSanitizer and/or valgrind, and to make sure nothing crashes. But a full test of functionality with specific semantic test is also required.

The compiled code isn't that big, about 3.6kB of code. And bit less on some embedded targets (~2.6kB).

```
$ gcc -Os -g0 -m32 -c -o pb_copy_gcc_x86_Os.o pb_copy.c
$ clang -Os -g0 -m32 -c -o pb_copy_clang_x86_Os.o pb_copy.c
$ clang -Oz -g0 -m32 -c -o pb_copy_clang_x86_Oz.o pb_copy.c
$ xtensa-lx106-elf-gcc -Os -g0 -c -o pb_copy_gcc_lx106.o pb_copy.c -I/usr/include/
$ arm-none-eabi-gcc -Os -g0 -c -o pb_copy_gcc_arm-none-eabi_Os.o pb_copy.c -I/usr/include/  # EABI5
$ arm-none-eabi-gcc -mthumb -Os -g0 -c -o pb_copy_gcc_thumb_arm-none-eabi_Os.o pb_copy.c -I/usr/include/  # EABI5
$ size pb_copy.o 
   text	   data	    bss	    dec	    hex	filename
   3643	      0	      0	   3643	    e3b	pb_copy_clang_x86_Os.o
   3575	      0	      0	   3575	    df7	pb_copy_clang_x86_Oz.o
   3616	      0	      0	   3616	    e20	pb_copy_gcc_x86_Os.o
   2677	      0	      0	   2677	    a75	pb_copy_gcc_lx106_Os.o
   3459	      0	      0	   3459	    d83	pb_copy_gcc_arm-none-eabi_Os.o
   2495	      0	      0	   2495	    9bf	pb_copy_gcc_thumb_arm-none-eabi_Os.o
```

`gcc version 10.2.1 20210110 (Debian 10.2.1-6)` , targeting 32-bit i686, but probably with SSE2 available.
`clang 11.0.1-2, targeting 32-bit i386`
`xtensa-lx106-elf-gcc (10.2.1-3+8) 10.2.1 20201224`
`arm-none-eabi-gcc (15:8-2019-q3-1+b1) 8.3.1 20190703 (release) [gcc-8-branch revision 273027]`

(Compiling at -O3 for amd64 gives about 4.1-5.1kB depending on a compiler, which is also quite OK).

This is with `PB_ENABLE_MALLOC`. Without `PB_ENABLE_MALLOC` it will be about a half of that.

Obviously, if not used, linker should be smart enough to GC it out of the executable.

baryluk, 2021-01-18:
Just looked a bit more into `fuzztest.c`. Hooking `pb_deep_copy` and `pb_deep_merge` into it would really trivial from what I see, for example in `do_roundtrip`. Also `run_iteration` could be easily extended to generate two random messages, and do a merge (or replacement) between them, to also test it. That also looks very easy to do.

I noticed there are also callback decodes. I am not sure yet how to handle these. I don't use them.

PetteriAimonen, 2021-01-19:
I think the most reasonable thing for callbacks is just to skip over them. They are very application-specific so I don't think there is a reasonable way to copy between them by generic code.

ipsod, 2022-02-19:
@baryluk I was looking for merge functionality and found this issue.  How is your work on this going?