nanopb/nanopb #174: Encode callback is called twice

oije2333, 2015-10-07:
When encoding a message containing a sub-message that has a string field, the encode callback is called twice.

Modified the "simple" example to demonstrate:

simple.proto:

    // A very simple protocol definition, consisting of only
    // one message.

    syntax = "proto2";

    message SimpleMessage {
        optional string lucky_number = 1;
    }

    message Response {
      optional SimpleMessage simpleMessage = 1;
    }

simple.c:

    #include <stdio.h>
    #include <pb_encode.h>
    #include <pb_decode.h>
    #include "simple.pb.h"

    bool encode_string(pb_ostream_t *stream, const pb_field_t *field, void * const *arg)
    {
        printf("encode_string...\n");

        char *str = "Hello world!";
    
        if (!pb_encode_tag_for_field(stream, field))
            return false;
    
        return pb_encode_string(stream, (uint8_t*)str, strlen(str));
    }

    bool decode_string(pb_istream_t *stream, const pb_field_t *field, void **arg)
    {
        printf("decode_string...\n");

        uint8_t buffer[1024] = {0};
    
        /* We could read block-by-block to avoid the large buffer... */
        if (stream->bytes_left > sizeof(buffer) - 1)
            return false;
    
        if (!pb_read(stream, buffer, stream->bytes_left))
            return false;
    
        /* Print the string, in format comparable with protoc --decode.
         * Format comes from the arg defined in main().
         */
        printf((char*)*arg, buffer);
        return true;
    }

    int main()
    {
        /* This is the buffer where we will store our message. */
        uint8_t buffer[128];
        size_t message_length;
        bool status;
    
        /* Encode our message */
        {
            /* Allocate space on the stack to store the message data.
             *
             * Nanopb generates simple struct definitions for all the messages.
             * - check out the contents of simple.pb.h!
             * It is a good idea to always initialize your structures
             * so that you do not have garbage data from RAM in there.
             */
            Response message = Response_init_zero;
            message.has_simpleMessage = true;
        
            /* Create a stream that will write to our buffer. */
            pb_ostream_t stream = pb_ostream_from_buffer(buffer, sizeof(buffer));
        
            /* Fill in the callback */
            message.simpleMessage.lucky_number.funcs.encode = &encode_string;
        
            /* Now we are ready to encode the message! */
            status = pb_encode(&stream, Response_fields, &message);
            message_length = stream.bytes_written;
        
            /* Then just check for any errors.. */
            if (!status)
            {
                printf("Encoding failed: %s\n", PB_GET_ERROR(&stream));
                return 1;
            }
        }
    
        /* Now we could transmit the message over network, store it in a file or
         * wrap it to a pigeon's leg.
         */

        /* But because we are lazy, we will just decode it immediately. */
    
        {
            /* Allocate space for the decoded message. */
            Response message = Response_init_zero;
        
            /* Create a stream that reads from the buffer. */
            pb_istream_t stream = pb_istream_from_buffer(buffer, message_length);
        
            /* Fill in the callback and arg */
            message.simpleMessage.lucky_number.funcs.decode = &decode_string;
            message.simpleMessage.lucky_number.arg = "stringvalue: \"%s\"\n";

            /* Now we are ready to decode the message. */
            status = pb_decode(&stream, Response_fields, &message);
        
            /* Check for errors... */
            if (!status)
            {
                printf("Decoding failed: %s\n", PB_GET_ERROR(&stream));
                return 1;
            }
        }
    
        return 0;
    }

PetteriAimonen, 2015-10-08:
It is normal that encoding callbacks are called multiple times. This is necessary to obtain the submessage size, which in protobuf format must be encoded before the submessage itself. This is documented in e.g. https://koti.kapsi.fi/jpa/nanopb/docs/concepts.html#encoding-callbacks

I'm not aware of any alternate approaches which would support encoding into streams. When encoding into memory buffers, the re-encoding would be possible to avoid. However such optimization has not been implemented so far. (Listed as a limitation in https://koti.kapsi.fi/jpa/nanopb/docs/index.html )

PetteriAimonen, 2015-10-24:
Behaviour appears to be as designed and as documented, so I'm closing this issue. Please reopen or comment if you think otherwise.

closed 2015-10-24 by PetteriAimonen