nanopb/nanopb #175: Mishandled callback field in sub-message inside oneof

oije2333, 2015-10-07:
If a oneof contains a sub-message that has a string field, the encode callback is called twice, and the decode callback is never called. Modified the "simple" example to demonstrate.

simple.proto:

    // A very simple protocol definition, consisting of only
    // one message.

    syntax = "proto2";

    message SimpleMessage {
        optional string lucky_number = 1;
    }

    message Response {
      oneof response {
       SimpleMessage simpleMessage = 1;
      }
    }

simple.c:

    #include <stdio.h>
    #include <pb_encode.h>
    #include <pb_decode.h>
    #include "simple.pb.h"

    bool encode_string(pb_ostream_t *stream, const pb_field_t *field, void * const *arg)
    {
        printf("encode_string...\n");

        char *str = "Hello world!";
    
        if (!pb_encode_tag_for_field(stream, field))
            return false;
    
        return pb_encode_string(stream, (uint8_t*)str, strlen(str));
    }

    bool decode_string(pb_istream_t *stream, const pb_field_t *field, void **arg)
    {
        printf("decode_string...\n");

        uint8_t buffer[1024] = {0};
    
        /* We could read block-by-block to avoid the large buffer... */
        if (stream->bytes_left > sizeof(buffer) - 1)
            return false;
    
        if (!pb_read(stream, buffer, stream->bytes_left))
            return false;
    
        /* Print the string, in format comparable with protoc --decode.
         * Format comes from the arg defined in main().
         */
        printf((char*)*arg, buffer);
        return true;
    }

    int main()
    {
        /* This is the buffer where we will store our message. */
        uint8_t buffer[128];
        size_t message_length;
        bool status;
    
        /* Encode our message */
        {
            /* Allocate space on the stack to store the message data.
             *
             * Nanopb generates simple struct definitions for all the messages.
             * - check out the contents of simple.pb.h!
             * It is a good idea to always initialize your structures
             * so that you do not have garbage data from RAM in there.
             */
            Response message = Response_init_zero;
            message.which_response = Response_simpleMessage_tag;
        
            /* Create a stream that will write to our buffer. */
            pb_ostream_t stream = pb_ostream_from_buffer(buffer, sizeof(buffer));
        
            /* Fill in the callback */
            message.response.simpleMessage.lucky_number.funcs.encode = &encode_string;
        
            /* Now we are ready to encode the message! */
            status = pb_encode(&stream, Response_fields, &message);
            message_length = stream.bytes_written;
        
            /* Then just check for any errors.. */
            if (!status)
            {
                printf("Encoding failed: %s\n", PB_GET_ERROR(&stream));
                return 1;
            }
        }
    
        /* Now we could transmit the message over network, store it in a file or
         * wrap it to a pigeon's leg.
         */

        /* But because we are lazy, we will just decode it immediately. */
    
        {
            /* Allocate space for the decoded message. */
            Response message = Response_init_zero;
            message.which_response = Response_simpleMessage_tag;

            /* Create a stream that reads from the buffer. */
            pb_istream_t stream = pb_istream_from_buffer(buffer, message_length);
        
            /* Fill in the callback and arg */
            message.response.simpleMessage.lucky_number.funcs.decode = &decode_string;
            message.response.simpleMessage.lucky_number.arg = "stringvalue: \"%s\"\n";

            /* Now we are ready to decode the message. */
            status = pb_decode(&stream, Response_fields, &message);
        
            /* Check for errors... */
            if (!status)
            {
                printf("Decoding failed: %s\n", PB_GET_ERROR(&stream));
                return 1;
            }
        }
    
        return 0;
    }

PetteriAimonen, 2015-10-08:
This is more of a documentation bug, but comes up very often lately. I do not see a good way to implement callbacks inside oneofs, but you can see e.g. here for workarounds:
https://groups.google.com/forum/#!topic/nanopb/WLO6AYO6QI0

In issue #148 it was considered whether the generator should error-out in this situation, but it seems more sensible that non-callback fields can still be used even if the message contains callbacks.

oije2333, 2015-10-08:
OK, for now I was able to get it working using the technique described in examples/using_union_messages. Code's just a little clunkier this way.

kamathba, 2015-11-11:
I can think of an implementation that could work using a lookup table, and would have a cost of one pointer per PB_FIELD but would remove pb_callback_t's from the decoded structs. It would also have the caveat that the decoded object couldn't move in memory, otherwise it would require re-assigning of all it's callbacks.

The jist would be use the address of the decoded message container of the callback_t to lookup a encode/arg and decode/arg. the basic struct would have those 5 pointers (address of containing message plus the normal callback_t fields). You could generate some macros to create a lookup table of size whatever and add/remove handles to/from it.

Seems like a lot of buts/caveats though, and I'm relatively confident I'm missing something big/obvious. Tricky problem.

PetteriAimonen, 2015-11-11:
While we are comparing approaches, I'll mention one that I've been thinking about:

1) When generator detects callbacks inside oneof, it will generate a "setup_oneof" callback field on the top level.
2) This setup_oneof will be called when the type of the oneof has been resolved. It will get as arguments the field descriptor, data pointer and a free arg field.
3) The setup_oneof function will set all callback fields in the message to be decoded, like:

    bool setup_oneof(const pb_field_t *field, void *data, void *arg)
    {
        if (field->tag == MyMessage_oneof1_tag)
        {
             MySubMessage *submsg = data;
             submsg->callback1.func = &whatever;
             ....
        }
    }

The second approach I've sometimes thought about is defining callbacks at generator time, e.g.

    repeated int32 myfield = 1 [(nanopb).callback = myfieldcallback];

But that still leaves the issue of passing context to the callback, so does not really solve this issue.

kamathba, 2015-11-11:
I like your first approach. Do you have a function for fully checking if a submessage/message contains a callback at any depth? I'm guessing that information is owned by the fields based on their types + options.

What if there are options defined that cause a message with a oneof to *sometimes* have callbacks? E.G. the oneof is in message bar, so bar.*my_oneof*, and my_oneof has a string, and something like this happens in your .options:

foo1.bar.my_oneof.my_string max_size:16

And then there is a foo2 that contains bar but doesn't specify the string max size. I guess its a harmless field when there are no callbacks anywhere within the oneof. Now that I think about it this could happen already in nanopb without oneofs so there must be rules for the situation where a submessage is used in more than one location and has different option in both. Does the whole message become a pb_callback_t if has non-deterministic size fields? 

You say in step 3 it could set the callbacks before decode, but for simplicity and uniform usage should it also set the encode callbacks/args before encode? There could be something messy/inflexible about it in the encode case.

PetteriAimonen, 2015-11-11:
Callbacks in encoding oneofs should work currently already,
so I think this is needed for decoder only. Not sure if there
might be some use for it in encoder or not.

The generator does not currently have such recursive check,
but it does have information about all the messages so it could
be implemented. In some multiple-file cases there can be problems,
but when #116 is finally resolved even those go away.

> foo1.bar.my_oneof.my_string max_size:16

I don't think this is allowed by the syntax. It is always
PackageName.MessageName.fieldname, where PackageName and MessageName
may have multiple parts but only as far as they are nested message 
definitions.

kamathba, 2015-11-11:
Yea, that makes more sense. I was combining types/field_instances in my thinking for a second.

Jeroen88, 2018-04-12:
@PetteriAimonen I am experimenting with your excellent nanopb using a oneof with callbacks. It seems that my code is encoding correctly, setting the right single callback and arg via the union, however decoding is not working completely. A callback is never called, because I do not know how to set it, after the type is decoded. I can not find the mentioned  setup_oneof func, has it been implemented? Or is another way available?

PetteriAimonen, 2018-04-13:
@Jeroen88 This issue is still open, there is no `setup_oneof()` so far.

The easiest option currently is to set `(nanopb).no_unions = true` which will generate a separate field for each oneof entry and supports callbacks.

PetteriAimonen, 2019-01-25:
Now there is a per-message callback mechanism that will work also with oneofs.
Currently not much documentation for it yet, but the network_server example has been updated.

simeonfelis, 2019-02-21:
Looking at the example it appears to me that

    ListFilesResponse_callback

Is a fixed function name given through the fileproto.options:

    ListFilesResponse.file  type:FT_CALLBACK, callback_datatype:"void*"

Please consider following: we received a full message (encoded), which contains a repeated fields:

    message Message {
      message KeyValue {
        int32 key = 1;
        int32 value = 2;
      }
      message Request {
        repeated KeyValue params = 10 [(nanopb).type=FT_CALLBACK,(nanopb).callback_datatyp="void*"]
      }
      oneof payload {
        Request         write = 100;
        Response      writeRsp = 200;
      }
    }

We now want to validate the kv-pairs. therefore I would like to decode the message and use a callback that just validates each kv-entry.

When validating succeeds, I want to decode again, this time with a callback that performs the actual write actions.

I would then need to put the desired action (validating/writing) into the stream state, I guess?

PetteriAimonen, 2019-02-21:
Hmm yeah, easiest would be global variable, but that is of course often not desirable. Without oneofs, you could use `callback_datatype="struct {bool validate_only; void* data; }"`, but that is again a problem here. Stream state works if you are already using a custom stream.

Perhaps one way would be if there was a way to add extra non-protobuf fields to the generated struct. For now it could be simulated by adding `bool validate_only = 9999;` to the `Message`. Then that can be accessed as `((Message*)((char*)field->message - offsetof(Message, payload.write)))->validate_only`. The pointer arithmetic could be made simpler by adding a `container_of(field->message, Message, payload.write)` macro like in Linux kernel.

Perhaps the original `setup_oneof()` callback might be useful also, though even then one would have to pass the information through many steps.

simeonfelis, 2019-02-21:
I just stick back to 

    message Message {
      option (nanopb_msgopt).no_unions = true;
    ....

Like this callbacks work for my use case (validate, read/write, generate reply, ride pony).

And hope my stack won't blow up. It's getting pretty narrow in there...

PetteriAimonen, 2019-02-21:
Ok.

I'm not sure I understand what you mean by things getting narrow, but if there is anything specific on the nanopb side that is important to your usecase, I'm interested to hear.

simeonfelis, 2019-02-22:
I just have a small stack size (narrow).

I'm not familiar with nanopb internals, and on the first sight the suggestions by you don't appeal to me (I wouldn't be surprised if on a second look your suggestions would be a minblowing, positive experience).

Our usecase is a bit specific:

1) we receive a websocket message containing a protobuffer encoded payload
2) the payload (raw data) is stored on heap or on a filesystem, depending on the payload size announced in the websocket frame

The payload is the analyzed

1) The raw data is decoded with no callbacks to find out what's inside the oneof field 'payload' 
2) The corresponding handler, eg. writeReq is called, with the received, raw data
3) writeReq decodes the raw data multiple time:
3.1) the repeated KeyValue is decoded with a validator callback
3.2) the validation status is stored in the stream state
3.3) when validation succeeds, the repeated KeyValue is decoded with a writer callback
3.3) I've yet to find out how to build response holding a repeated KeyValue with all written parameters as confirmation

I tried to test the option from git master ((nanopb).type=FT_CALLBACK), however the library changes of *.c and *.h files were a bit too dramatic for me, and also my usecase requires a bit of special handling, where the fixed, single callback adds more complexity.

Maybe I'm doing something wrong, and there would be a more elegant approach.

matutter, 2019-07-04:
Wanted to check if there is any progress or plans for this issue or is [using_union_messages](https://github.com/nanopb/nanopb/blob/master/examples/using_union_messages/decode.c) still the best pattern to use?

@PetteriAimonen I like the `setup_oneof` callback idea. I was thinking it might be easier to just pass a substream to the callback and let the user handle the entire oneof message. 

```c
bool decode_my_union(pb_istream_s* stream, const pb_field_t* field, void* parent_struct, void** arg) {
  // User can check parent_struct.which_payload and continue to decode 
}
```

PetteriAimonen, 2019-12-18:
I ended up adding an option for message-level callback (like the `setup_oneof` idea) also.
There is a test case / example that demonstrates the use of both: https://github.com/nanopb/nanopb/tree/master/tests/oneof_callback

closed 2019-12-20 by PetteriAimonen