nanopb/nanopb #177: Segfault on decode of a message

dakoner, 2015-10-08:
Hi, I have a pretty simple message, which I call SensorDataRequest, that I've serialized.  Here is the ASCII representation:

timestamp_key: 42
interval {
  count: 1
  frequency: 20
}
pin {
  analog_pin {
    pin: 0
  }
}

and the corresponding hex bytes after encoding with the Python protobuf library.
08 2a 12 04 08 01 10 14 1a 04 52 02 08 00 
I am able to decode these hex bytes with the Python protobuf library.

However, when I try to decode this using nanopb, I get an segfault:

#include <stdlib.h>
#include <stdio.h>
#include <pb_decode.h>
#include "../nanopb/sensor.pb.h"
uint8_t data[] = {0x08, 0x2a, 0x12, 0x04, 0x08, 0x01, 0x10, 0x14, 0x1a, 0x04, 0x52, 0x02, 0x08, 0x00};

int main(void) {
  sensor_SensorDataRequest sdr;
  pb_istream_t stream = pb_istream_from_buffer(&data[0], sizeof(data));
  bool status = pb_decode(&stream, sensor_SensorDataRequest_fields, &sdr);

  return 0;
}

In case it helps, here is the valgrind output from parsing the bytes:
==30688== 
==30688== Conditional jump or move depends on uninitialised value(s)
==30688==    at 0x402561: decode_callback_field(pb_istream_s*, pb_wire_type_t, pb_field_iter_s*) (pb_decode.c:600)
==30688==    by 0x40274A: decode_field(pb_istream_s*, pb_wire_type_t, pb_field_iter_s*) (pb_decode.c:658)
==30688==    by 0x402D28: pb_decode_noinit (pb_decode.c:893)
==30688==    by 0x402E8E: pb_decode (pb_decode.c:934)
==30688==    by 0x40062F: main (pbdecode.cc:10)
==30688== 
==30688== Use of uninitialised value of size 8
==30688==    at 0x4025C4: decode_callback_field(pb_istream_s*, pb_wire_type_t, pb_field_iter_s*) (pb_decode.c:612)
==30688==    by 0x40274A: decode_field(pb_istream_s*, pb_wire_type_t, pb_field_iter_s*) (pb_decode.c:658)
==30688==    by 0x402D28: pb_decode_noinit (pb_decode.c:893)
==30688==    by 0x402E8E: pb_decode (pb_decode.c:934)
==30688==    by 0x40062F: main (pbdecode.cc:10)
==30688== 

Also, in case it helps, here's the .proto definition. Note that I'm using a oneof,
syntax = "proto2";

package sensor;
message AnalogPin {
  required int32 pin = 1;
};
message DigitalPin {
  required int32 pin = 1;
};
message VirtualPin {
  required int32 pin = 1;
};
message Pin {
  oneof pin {
    AnalogPin analog_pin = 10;
    DigitalPin digital_pin = 11;
    VirtualPin virtual_pin = 12;
  }
};
message Interval {
  required int32 count = 1;
  optional int32 frequency = 2;
};
message SensorDataRequest {
  required uint32 timestamp_key = 1;
  required Interval interval = 2;
  repeated Pin pin = 3;
};

PetteriAimonen, 2015-10-08:
Try initializing the structure to zeros:

sensor_SensorDataRequest sdr = {};

You have a callback field there, so now it calls random function pointer because the value is uninitialized.

dakoner, 2015-10-08:
Thanks for the quick response!  That worked.  In retrospect, I should have realized there was a callback pointer being followed, from the valgrind output.

closed 2015-10-08 by dakoner