nanopb/nanopb #178: Support Code Generator Plugin Insertion Points

tullisar, 2015-10-09:
It would be useful for nanopb to have insertion points (as described in plugin.proto) in it's generated code, so that other code generator plugins could add additional target specific code where necessary. A couple useful locations that I think of:

Header files (.pb.h):
- Include scope (after includes that nanopb adds)
- Pre and post-type declarations (prior to all enum/structs nanopb creates)
- Message struct scope (within the scope of a struct created by nanopb, after the fields)

Source file (.pb.c)
- Include scope (after includes that nanopb adds)
- Pre and post-field descriptors (prior to the field descriptor definitions)

PetteriAimonen, 2016-02-06:
I have now added insertion points at:
1. include directives
2. every message struct
3. end of file

I do not wish to fill the files with these without need, though. So more points can be added but will need a specific use case and a separate issue report.

PetteriAimonen, 2016-02-13:
Feature released in nanopb-0.3.5.

closed 2016-02-13 by PetteriAimonen