nanopb/nanopb #180: Field value not always set on AVR platform

bverhoeven, 2015-10-12:
I'm interested in using nanopb on an atmega328p (AVR) using avr-gcc/avr-g++ (4.8.0). I have created a small test project to experiment nanopb but I'm seeing unexpected behavior and I hope someone can tell me what I'm doing wrong or what's happening.

I have trimmed down the protocol definition to:

```protobuf
message TestMessage
{
        enum Type
        {
                ABC = 1;
        }

        optional Type type = 1;
        optional fixed32 length = 2;
}
```

And have created the following test code, based on the `simple` example.

```c
uint8_t buffer[128] = { 0 };

void test_nanopb()
{
    pb_ostream_t pb_stream = pb_ostream_from_buffer(buffer, sizeof(buffer));

    TestMessage message = TestMessage_init_default;
    message.has_length = true;
    message.length = 8;

    if (pb_encode(&pb_stream, TestMessage_fields, &message))
    {
        size_t i;
        for(i = 0; i < pb_stream.bytes_written; i++)
        {
            putchar(buffer[i]);
        }
    }

    // ...
}
```

When testing this code on Linux (g++ 4.8.4) the output of this program is: ```15 *08* 00 00 00```, which is correct. However when testing the same code on the atmega/AVR the output is: ```15 *00* 00 00 00``` which is missing the value of the `length` field. I do not understand why this is happening.

The `type` field seems to be involved: If I remove the `type` field (and keep the `length` field and enum definition) then both programs output the same value: ```15 08 00 00 00```. I have tried setting a default value for the enum and have also tried to change the `fixed32` to `int32`, without any luck.

I have tested with nanopb from git (0.3.5-dev, master) and with the last stable release (0.3.4). Both are showing the same issue.

Any help is appreciated.

PetteriAimonen, 2015-10-12:
Hmm, tried to reproduce the problem under simulator with avr-gcc 4.8.1, but I get the correct output (15 08 00 00 00). What compilation options are you using?

If you can test a few things it could help narrow down the problem:

1) After encoding, print out the structure byte-by-byte:
for (i = 0; i < sizeof(message); i++) putchar(((uint8_t*)&message)[i]);
This will tell me the padding used by the compiler, and it will also show if something accidentally overwrites the value.

2) Upload your compiled main.elf file. I'll check manually if the field descriptors seem correct.

3) Are you using C++ or plain C for your main file? Shouldn't really matter, but if you are using C++ and if you can easily try C, it could give some hints.

4) Check if nanopb calculates the correct pointer to the field:
```
#include <pb_common.h>

...

pb_field_iter_t iter;
pb_field_iter_begin(&iter, TestMessage_fields, &message);
pb_field_iter_next(&iter);
printf("field %d pData is: %08x\n", iter->pos->tag, (unsigned)iter->pData);
printf("should be: %08x\n", (unsigned)&message.length);
```

bverhoeven, 2015-10-12:
Thanks for the quick response and sorry, the issue appears to be on my end.

I'm using avr-g++ for the project. Although i'm not using any C++ code at the moment. Compilation options are mostly taken from an older project, here's a list of all the options I'm currently using:

```
CPPFLAGS = -c -Os -mmcu=atmega328p -Wno-write-strings -Wall -fpack-struct
CPPFLAGS+= -fshort-enums -ffunction-sections -fdata-sections -funsigned-char
CPPFLAGS+= -fno-align-loops -fno-move-loop-invariants -fno-tree-loop-ivcanon
CPPFLAGS+= -funsigned-bitfields -D F_CPU=8000000
CPPFLAGS+= -MMD -MP
```

They're quite a lot so I've removed a few and found that the problem disappeared. Sorry for not looking at that earlier. I have narrowed the issue down to the `-fshort-enums` option which we (apparently) use to make an enum as small as the smallest integer that has enough space. With this option set I am seeing the issues as described, without it the behavior is as expected.

As I had already collected some of the diagnostics, here they are with `-fshort-enums` set:

* Output: ```15 00 00 00 00```
* Output of the structure: ```00 00 01 08 00 00 00```
* Pointer' check: ```field 2 pData is: 000008fb should be: 000008fa```

And without `-fshort-enums` set:

* Output: ```15 08 00 00 00```
* Output of the structure: ```00 00 00 01 08 00 00 00```
* Pointer' check: ```field 2 pData is: 000008fa should be: 000008fa```

The manual does warn that the switch causes GCC to generate code that is not binary compatible with code generated without that switch so I guess that is what is happening right now :-(

closed 2015-10-12 by bverhoeven

PetteriAimonen, 2015-10-13:
No problem, good that the issue got resolved.

This is not the first time I've seen bugs due to different struct layout between C++ and C code.. it would be great if there was a way to assert that sizeof() is the same between .pb.c and wherever .pb.h is included.