nanopb/nanopb #182: Suppression of has_* fields for protobuf 3

vvanbeveren, 2015-10-16:
In protobuf3 all plain fields are optional. This generates a lot of has_* fields, which is not desirable in resource constraint environments such as microcontrollers. A way to suppress, or limit, the generation of has_* for protobuf3 would mean that precious resources are not wasted.

kylemanna, 2015-10-16:
How would you know if the field was ever set without the `has_*` field in C without metadata like this?

PetteriAimonen, 2015-10-17:
In proto3, the default value is always zero-ish:
https://developers.google.com/protocol-buffers/docs/proto3#default

So just have to check for those conditions.

Another issue is how to convey to nanopb runtime that we are in proto3 
mode. This should probably be a per-message info somewhere.
I would really like to be able to reuse PB_HTYPE_REQUIRED etc. for 
different purposes in proto3.

PetteriAimonen, 2016-10-23:
Fixed by #216.

PetteriAimonen, 2016-10-30:
Fix released in nanopb-0.3.7

closed 2016-10-30 by PetteriAimonen