nanopb/nanopb #184: Allow anonymous unions for oneof

kamathba, 2015-11-04:
Would anybody be interested in a way to specify through the .options file that a specific union/message field be an anonymous union/structure? It could just be my own bad design patterns but it seems like something that could be handy to prevent Response.FooResponse.foo_bar_response type field names.

It would maybe be more useful for unions, and would be limited in application to fields only, not native message types.

I might be missing something that would make this tricky but I feel like I could have cleaner generated files if this was an option.

PetteriAimonen, 2015-11-05:
Shouldn't be too problematic to implement, and there is a similar
enum_short_names option also to clean up the generated names. So
yeah, would make sense to have some anonymous_unions generator
option.

PetteriAimonen, 2015-11-14:
Implemented by pull request #186.

PetteriAimonen, 2016-02-13:
Feature released in nanopb-0.3.5.

closed 2016-02-13 by PetteriAimonen