nanopb/nanopb #188: protoc-gen-nanopb crashes with oneof of message and enum

mozturk, 2015-11-20:
Using the binary release of version 0.3.4, the code generator crashes when compiling the following protobuf defintion:

```
syntax = "proto2";

message MessageOne
{
    required uint32 one = 1;
    required uint32 two = 2;
    required uint32 three = 3;
    required int32 four = 4;
}

enum EnumTwo
{
    SOME_ENUM_1 = 1;
    SOME_ENUM_2 = 5;
    SOME_ENUM_3 = 6;
    SOME_ENUM_4 = 9;
    SOME_ENUM_5 = 10;
    SOME_ENUM_6 = 12;
    SOME_ENUM_7 = 39;  
    SOME_ENUM_8 = 401;
}

message OneofMessage
{
    oneof payload {
        MessageOne message = 1;
        EnumTwo enum = 2;
    }
}
```

The code generator crashes with the following error:
```
Traceback (most recent call last):
  File "<string>", line 6, in <module>
  File "__main__.py", line 128, in <module>
  File "__main__protoc-gen-nanopb__.py", line 1497, in <module>
  File "__main__protoc-gen-nanopb__.py", line 1482, in main_plugin
  File "__main__protoc-gen-nanopb__.py", line 1401, in process_file
  File "__main__protoc-gen-nanopb__.py", line 1062, in generate_header
  File "__main__protoc-gen-nanopb__.py", line 836, in encoded_size
  File "__main__protoc-gen-nanopb__.py", line 686, in encoded_size
AttributeError: 'int' object has no attribute 'symbols'
--nanopb_out: protoc-gen-nanopb: Plugin failed with status code 1.
```

Note that if I change the `oneof` field to contain two `MessageOne` fields instead of a `MessageOne` and `EnumTwo` field it does compile fine.

PetteriAimonen, 2015-11-20:
Yeah, bug is reproducible.

PetteriAimonen, 2016-02-13:
Fix released in nanopb-0.3.5.

closed 2016-02-13 by PetteriAimonen