nanopb/nanopb #189: Missing `Message_size` when `oneof` exists

jameslan, 2015-12-14:
We use `oneof` in the message to save memory using the union of sub messages. Unfortunately in this case the `Message_size` won't be generated  so that we don't have an upper limit of encoded message

PetteriAimonen, 2015-12-14:
Can you provide the .proto file that shows this problem?

jameslan, 2015-12-16:
Hi I think I found the real issue I met:

When using sub message with `oneof` and the sub message is imported and the option is defined on the sub message, the Message_size marco of outer message wont be generated.

For example,

Foo.proto
```protobuf
syntax = "proto2";

import "Bar.proto";

message Foo {
  oneof value {
    Bar bar = 1;
  }
}
```

Bar.proto
```protobuf
syntax = "proto2";

message Bar {
  optional bytes data = 1;
}
```

Bar.options
```
Bar.data max_size: 128
```

`Foo_size` won't be generated in Foo.pb.h while `Bar_size` will be in Bar.pb.h

PetteriAimonen, 2015-12-16:
Yeah, there are some issues related to .options and imported messages. 
Most importantly issues #116 and #172.

If your files are in different folders, try giving the include path to 
nanopb also like this: "--nanopb_out=-Isubdir:."

To fix this properly requires some upstream changes to protoc, which I 
haven't had time to do as the first version of a patch didn't go 
through.

PetteriAimonen, 2019-12-20:
I'll close this off even though the underlying complexity still exists.

In 0.4.0, `nanopb_generator.py` now accepts multiple `.proto` files on command line at once, which ensures that dependencies are found correctly.

closed 2019-12-20 by PetteriAimonen