nanopb/nanopb #192: Default strings could be a bit smaller

rsaxvc, 2016-02-11:
**Steps to reproduce the issue**

1. checkout 6f0dfe1b3f
2. build client/server example

**What happens?**
nanopb/examples/network_server/fileproto.pb.c has:
const char ListFilesRequest_path_default[128] = "/";

**What should happen?**
I think it could save 126 bytes of space by moving to:
const char ListFilesRequest_path_default[2] = "/";
or
const char ListFilesRequest_path_default[] = "/";

PetteriAimonen, 2016-02-12:
Hmm, true. Though this will require a separate path in here: https://github.com/nanopb/nanopb/blob/master/pb_decode.c#L785
Currently it just memcpy()'s the all, and reading from uninitialized memory is not a good idea.

With #114 and the default value differences in protobuf 3, I'm not sure if this will be relevant anymore. But it's a reasonable thing to keep in mind when implementing the other changes, so I'm keeping this open.

PetteriAimonen, 2019-01-25:
Now the whole message default values are encoded in protobuf format, which uses only the actual length of string for storage. Additionally it saves a bit of space by not having to store separate pointers for each field.

closed 2019-12-20 by PetteriAimonen