nanopb/nanopb #195: Message size not calculated if a submessage includes bytes

simonhaines, 2016-03-04:
**Steps to reproduce the issue**

Compile the following .proto file:
```
message TestMessage {
	required uint32 id = 1;
	required bytes payload = 2;
}
message EncapsulatedMessage {
	required uint32 id = 1;
	required TestMessage test = 2;
}
```

(On Windows)
```
> nanopb-0.3.5-windows-x86\generator-bin\protoc.exe --nanopb_out=. test.proto
```

The header file contains a calculated message size that references the undefined size of TestMessage:
```
/* Maximum encoded size of messages (where known) */
#define EncapsulatedMessage_size                 (12 + TestMessage_size)
```

**What happens?**
The header file fails to compile.
```
test.pb.h:50:56: error: 'TestMessage_size' undeclared here (not in a function)
```

This may be related to #189 and #172 but I'm not using options to specify a size. I'm using the callback interface.

PetteriAimonen, 2016-03-04:
The callback can write any amount of data, thus it is not possible to calculate the size for the TestMessage.

I think the #define should not cause a compilation error as long as you don't use it. Though it is clearly a bug also, as it shouldn't generate the #define at all.

simonhaines, 2016-03-04:
OK, I will migrate to the options interface to specify a maximum size.

PetteriAimonen, 2016-03-04:
Nah, let's keep this open because there is still the bug of a not-working #define being generated.

Snaipe, 2016-03-04:
@PetteriAimonen You could `#define` the size of the message to be `-1`, and prepend the definition with `_Pragma` (or `__pragma` on msvc) when available to tell the user that the size cannot be computed as a compiler warning. It's not exactly a compiler error, but at least it's more explicit than failing with a possibly obscure message from the compiler.

PetteriAimonen, 2016-03-04:
@Snaipe To me this seems just a bug in the generator. You can see that it does not generate any #define for the TestMessage. It should similarly omit it for EncapsulatedMessage.

As for informing the user, I'm kind of hoping that the "(when known)" comment is enough of a hint ;)

PetteriAimonen, 2016-03-04:
Hmm yeah, actually not so easy to avoid generating the #define as that would conflict with #172. So instead just added comments like /* TestMessage_size depends on runtime parameters */ to make it a bit easier to understand.

PetteriAimonen, 2016-06-19:
Fix released in nanopb-0.3.6.

closed 2016-06-19 by PetteriAimonen

bonastreyair, 2020-09-15:
I found a way `#defines variable_*_size` are still generated even when they should be unknown. So I can't compile it either. To  solve it I need to comment all those "#define and unions" that should not exist.

I have like 2 levels nested messages with `oneof` and with repeated values, that's what it happens...

## Example
### main.proto file
```
syntax = "proto3";

package message;

import "type.proto";

message BigMessage {
  /* Only one of this message can be used. */
  message Message {
    oneof message {
      type.MessageType1 type1 = 1; 
      type.MessageType2 type2 = 2;
    }
  }
  repeated Message message = 1;
}
```

### type.proto file
```
syntax = "proto3";

package type;

message MessageType1 {
  message Number {
    oneof number {
      int32 num1 = 1;
      float num2 = 2;
    }
  }
  repeated Number number = 1;
}

message MessageType2 {
  message OtherNumber {
    oneof otherNumber {
      int32 num1 = 1;
      float num2 = 2;
      string num3 = 3;
    }
  }
  repeated OtherNumber number = 1;
}
```

Then in `main.pb.h` it appears:
```
/* Maximum encoded size of messages (where known) */
/* message_BigMessage_size depends on runtime parameters */
union message_BigMessage_Message_message_size_union {char f1[(6 + type_MessageType1_size)]; char f2[(6 + type_MessageType2_size)];};
#define message_BigMessage_Message_size          (0 + sizeof(message_BigMessage_Message_message_size_union))
```

and int `type.pb.h` appears:
```
/* Maximum encoded size of messages (where known) */
/* type_MessageType1_size depends on runtime parameters */
#define type_MessageType1_Number_size            11
/* type_MessageType2_size depends on runtime parameters */
/* type_MessageType2_OtherNumber_size depends on runtime parameters */
```

So when I import main.pb.h it tells me that `type_MessageType1_size` and `type_MessageType2_size` are not defined and I can't compile.

bonastreyair, 2020-09-15:
I just found that 5 days ago with https://github.com/nanopb/nanopb/commit/ca01b338e9a9f52e88d9e18c47faf2956f59ea47 commit you adressed this issue, I'll try if this works and let you know. 😄 

Any idea on when will you release `0.4.3` with all latest fixes?

PetteriAimonen, 2020-09-16:
@bonastreyair I hope to release it soonish (within a month or so), but it depends a lot on how much time I have available.