nanopb/nanopb #197: Missing required field will not fire any error

vladaman, 2016-03-23:
When field is **required** the code will not fire any error if values are not set.

pb_encode returns still true and stream.errmsg is null

**protobuf:**
```protobuf
message sensor_measurement {
  required int32 sensor_id = 1;
  required float current_value = 2;
}
```
**C code:**
```c
	sensor_measurement mdata = sensor_measurement_init_default;
	mdata.sensor_id = 12;
	// mdata.current_value = 3.33;
	stream = pb_ostream_from_buffer(buffer, sizeof(buffer));
	status = pb_encode(&stream, sensor_measurement_fields, &mdata);
	message_length = stream.bytes_written;

	printf("Error: %s -> %u\n",stream.errmsg , status);
	// Prints:
	// Error: (null) -> 1
	// Bytes out: 080c15000000004000
```

vladaman, 2016-03-23:
I propose a fix to change following line in pb_encode.c from:
```c
case PB_HTYPE_REQUIRED:
            if (!pData)
```
to:
```c
case PB_HTYPE_REQUIRED:
            if (!pData || pData == NULL || strlen(pData) == 0)
```

PetteriAimonen, 2016-03-24:
> When field is **required** the code will not fire any error if values are not set.

There is no attempt to error about missing fields when encoding.
For most types, such as int and float in your example, it is not even 
possible without requiring a separate has_ field for every required 
field.

> if (!pData || pData == NULL || strlen(pData) == 0)

I'm not sure how this is supposed to work. "!pData" and "pData == NULL" 
are equivalent conditions. And strlen(pData) does not make sense for 
e.g. integers. For example if your int32 field has value 1024, it would 
fail here.

PetteriAimonen, 2016-03-30:
Closing this off, as it does not seem like an issue worth fixing.

closed 2016-03-30 by PetteriAimonen