nanopb/nanopb #198: nanopb on ARM using Gcc does not pack repeated fields

vladaman, 2016-03-24:
Using _repeated_ field works well when compiled on PC with gcc. However when compiled for ARM and running on ARM it does not seem to pack any of these fields into a structure. They are just missing.

PetteriAimonen, 2016-03-24:
Please provide a runnable example code that reproduces the problem.

If that is not possible, provide:
1) .proto you are using
2) Set breakpoint right before pb_encode() call and give printout of 
the message structure.
3) Set breakpoint after pb_encode() call and give printout of the output 
data.

vladaman, 2016-03-29:
Here is the example

Protobuf message:
```protobuf
message Msg {
  required MessageType type = 1;
  required string device_id = 2;
  required int32 request_code = 3;
  required int32 sw_version = 4;
  repeated bytes data = 5; // 64 bytes chunks
  optional int32 clock_ts = 6;
  optional bytes signature = 7;
  optional bytes iv = 8; // AES Vector
}
```
C code:
```c
typedef struct _Msg {
    MessageType type;
    char device_id[16];
    long request_code;
    long sw_version;
    pb_size_t data_count;
    Msg_data_t data[32];
    bool has_clock_ts;
    long clock_ts;
    bool has_signature;
    Msg_signature_t signature;
    bool has_iv;
    Msg_iv_t iv;
} Msg;
 
message.type = MessageType_sensor_data_post_t;
sprintf(message.device_id,"867273022906916");
message.request_code = 0x12345678;
message.sw_version = 1;
message.data_count = 1;
 
for(u8 i=0;i<64;i++)buffer_d[i]=i;
message.data[0].size = 64;
memcpy(message.data[0].bytes,buffer_d,64);
 
status = pb_encode(&stream, Msg_fields, &message);
message_length = stream.bytes_written; // returns 27 bytes
```
Before encode:
```
09383637 32373330 32323930 36393136 00000000
78563412 01000000 01400001 02030405 06070809
0A0B0C0D 0E0F1011 12131415 16171819 1A1B1C1D
1E1F2021 22232425 26272829 2A2B2C2D 2E2F3031
32333435 36373839 3A3B3C3D 3E3F0000 00000000
00000000 00000000 00000000 00000000 00000000
00000000 00000000 00000000 00000000 00000000
00000000 00000000 00000000 00000000 00000000
00000000 00000000 00000000 00000000 00000000
```

After encode:
```
0809120F 38363732 37333032 32393036 39313618
F8ACD191 01200100
```

PetteriAimonen, 2016-03-29:
How do you initialize the stream?

PetteriAimonen, 2016-03-29:
(And same question for declaring buffer_d etc.; I would really prefer a *runnable* example code, not just a snippet.)

vladaman, 2016-03-29:
Hi Petteri,

Sorry for that. Here is more incl the declarations.
```c
void write_sensor_data(char* _imeicode,u16 _volt,u16 _temper,uint8_t *buffer,uint32_t _len)
{

uint8_t i;
size_t message_length;
bool status;
uint8_t buffer_d[256];//,buffer_m1[32],buffer_m2[32];
memset(buffer_d,0,256);
Msg message = Msg_init_default;

pb_ostream_t stream = pb_ostream_from_buffer(buffer, _len);

message.type = MessageType_sensor_data_post_t;

sprintf(message.device_id,"867273022906916");
message.request_code = 0x12345678;
message.sw_version = 1;
message.data_count = 1;

for(u8 i=0;i<64;i++)buffer_d[i]=i;
message.data[0].size = 64;
memcpy(message.data[0].bytes,buffer_d,64);

status = pb_encode(&stream, Msg_fields, &message);
message_length = stream.bytes_written; // returns 27 bytes as per dump above
}
```

PetteriAimonen, 2016-03-29:
So is the status successful, or is there some error message in the stream?

Other than that, you'll probably have to debug it further yourself. A good start may be setting a breakpoint in `encode_array()` function in pb_encode.c, and then stepping line-by-line to see what happens.

SMSoftwareCZ, 2016-03-30:
Hi, I am working with vladaman.
I try debug function encode_array() and there are the count =0, due this are retuned immediately with TRUE status without encoding.
Here is function call with pSize=0 inside the function encode_basic_field() :
` if (!encode_array(stream, field, pData, *(const pb_size_t*)pSize, func))`

PetteriAimonen, 2016-03-30:
Ok. Next step would be to check what is the value of the pSize pointer, and compare it against the value of "&message.data_count" pointer.

This is starting to seem like your .pb.c file and the main .c file are somehow compiled differently. Common causes would be having some old version of .pb.c in a different directory, or using incompatible C++ and C compilers for the different files.

SMSoftwareCZ, 2016-03-30:
Yes, there are differencies in pointer address (0x100???).
pSize = 0x2000f76c
&message.data_count = 0x2000f66c

SMSoftwareCZ, 2016-03-30:
OK, somewhere in function encode_basic_field() is error with computing the pointer to the array.

PetteriAimonen, 2016-03-30:
The computation logic is pretty well tested, so most likely is that your .pb.h and the compiled .pb.c do not match somehow.

If you can get a binary dump (memory dump) of 1) the Msg_fields array and 2) the msg structure, I can take a look.

SMSoftwareCZ, 2016-03-30:
Message memmrory dumb and structure are in previous posts..

SMSoftwareCZ, 2016-03-30:
The array strted here:
01400001 02030405 06070809
0A0B0C0D 0E0F1011 12131415 16171819 1A1B1C1D
1E1F2021 22232425 26272829 2A2B2C2D 2E2F3031
32333435 36373839 3A3B3C3D 3E3F

PetteriAimonen, 2016-03-30:
No, I mean the variables "Msg_fields" and "msg". And complete dumps of both; you can attach files if it is too long to put as text.

SMSoftwareCZ, 2016-03-30:
const pb_field_t Msg_fields[9] = {
    PB_FIELD(  1, ENUM    , REQUIRED, STATIC  , FIRST, Msg, type, type, 0),
    PB_FIELD(  2, STRING  , REQUIRED, STATIC  , OTHER, Msg, device_id, type, 0),
    PB_FIELD(  3, INT32   , REQUIRED, STATIC  , OTHER, Msg, request_code, device_id, 0),
    PB_FIELD(  4, INT32   , REQUIRED, STATIC  , OTHER, Msg, sw_version, request_code, 0),
    PB_FIELD(  5, BYTES   , REPEATED, STATIC  , OTHER, Msg, data, sw_version, 0),
    PB_FIELD(  6, INT32   , OPTIONAL, STATIC  , OTHER, Msg, clock_ts, data, 0),
    PB_FIELD(  7, BYTES   , OPTIONAL, STATIC  , OTHER, Msg, signature, clock_ts, 0),
    PB_FIELD(  8, BYTES   , OPTIONAL, STATIC  , OTHER, Msg, iv, signature, 0),
    PB_LAST_FIELD
};
typedef struct _Msg {
    MessageType type;
    char device_id[16];
    long request_code;
    long sw_version;
    pb_size_t data_count;
    Msg_data_t data[32];
    bool has_clock_ts;
    long clock_ts;
    bool has_signature;
    Msg_signature_t signature;
    bool has_iv;
    Msg_iv_t iv;
} Msg;

PetteriAimonen, 2016-03-30:
No, I need a memory dump from a debugger instead of the source code.

SMSoftwareCZ, 2016-03-30:
But the bytes what I sent you in older post are the dump from debuger..
again in txt files:
[message.txt](https://github.com/nanopb/nanopb/files/195622/message.txt)
[buffer.txt](https://github.com/nanopb/nanopb/files/195624/buffer.txt)

PetteriAimonen, 2016-03-30:
Yes, those are memory dumps like I want. But those are the variables "buffer_d" and "buffer".

To debug this issue, I need dumps of variables "Msg_fields" and "msg".

SMSoftwareCZ, 2016-03-30:
Msg_fields line in debuger: size =1 ?
01000000 01000000 00000206 00001000 00000000

SMSoftwareCZ, 2016-03-30:
"msg" ? You mena "Msg" (Msg message = Msg_init_default;) or what?

PetteriAimonen, 2016-03-30:
Nah, it is longer than that; your debugger is not showing it fully for some reason.

SMSoftwareCZ, 2016-03-30:
I can show more bytes in debuger  but I don´t know how much can be..?

SMSoftwareCZ, 2016-03-30:
More bytes from Msg_fields
[msg.txt](https://github.com/nanopb/nanopb/files/195654/msg.txt)

PetteriAimonen, 2016-03-30:
Ah sorry, yeah, I meant "message" not "msg".

The Msg_fields array should be 90 bytes long.

PetteriAimonen, 2016-03-30:
Yeah, that is enough.

SMSoftwareCZ, 2016-03-30:
OK, message bytes are in message.txt

PetteriAimonen, 2016-03-30:
Hmm I have checked the field array vs. message structure and they match, so it is not compilation issue.

Have you checked whether you have enough stack space available? Stack overflow could possibly cause this.

If it is not a stack issue, can you set a breakpoint in pb_common.c line 74 (return true) and take note of the iter->pData and iter->pSize values after each field?

SMSoftwareCZ, 2016-03-30:
pb_common.c line 74 
[iter.txt](https://github.com/nanopb/nanopb/files/195709/iter.txt)

SMSoftwareCZ, 2016-03-30:
line73: size_offset = 255 in array? due this the count = 0.

PetteriAimonen, 2016-03-30:
Hmm, so the addition is somehow getting messed up.

iter->pSize = (char*)iter->pData + iter->pos->size_offset;

And because pData is 0xf66d, and iter->pos->size_offset should be -1, the result should be 0xf66c. But instead it thinks iter->pos->size_offset is 255.

This in turn seems like pb_ssize_t would be unsigned, not signed. But it should be int8_t so it should be signed.

Which compiler are you using? Does it come with stdint.h or do you have to use a custom one?

SMSoftwareCZ, 2016-03-30:
It is GCC ARM Embedded 4.9-2015-q3
size definition:
#define PB_SIZE_MAX ((uint8_t)-1)
    typedef unsigned char pb_size_t;
    typedef char pb_ssize_t;
#endif

Snaipe, 2016-03-30:
Just as a heads up, the signedness of `char` isn't defined, pb_ssize_t could be either signed or unsigned here, which is most likely your issue.

PetteriAimonen, 2016-03-30:
Where does that define come from? The original in pb.h is like this:
    typedef uint_least8_t pb_size_t;
    typedef int_least8_t pb_ssize_t;

On GCC ARM Embedded, char is unsigned, so int8_t should be "signed char".

SMSoftwareCZ, 2016-03-30:
Yes, I try  int8_t and it looks fine.  
I thought  that char is signed..

PetteriAimonen, 2016-03-30:
In general, I don't recommend modifying pb.h. I think on GCC ARM Embedded, it should work without modifications. Other platforms, such as Windows, you can use pb_syshdr.h from the extra/ directory.

PetteriAimonen, 2016-03-30:
Glad that this is now resolved, though :)

closed 2016-03-30 by PetteriAimonen

SMSoftwareCZ, 2016-03-30:
OK, but for example int_least8_t are not defined and due this I must defined own typedef..

PetteriAimonen, 2016-03-30:
Hmm, it should be defined on every C99 compatible compiler. It is defined on my GCC ARM Embedded (4.9.3, Q2 2015).

On platforms that only support the 27 years old C89 standard, you can use pb_syshdr.h to do the typedefines. But GCC ARM Embedded shouldn't need this, as it has stdint.h which should define those.