nanopb/nanopb #199: C naming conversion for struct and enum names

philips77, 2016-04-27:
Hi,
Right now, the generated code does not change the proto CamelCase into typical C naming style.
Is there an option to enabled it somewhere? Or do you plan to add this future?
If I have a single proto file and generate a code to C or Java it always look very ugly in one of the languages. Here https://developers.google.com/protocol-buffers/docs/style they say how fields should look like, but imho message names should be converted from Camel Case into C standard.

PetteriAimonen, 2016-04-27:
No, currently there is no such option and I haven't had plans to do this. But this seems like a reasonable feature request, though I'm not going to work on it myself in the near future.

M1cha, 2020-05-28:
additionally, there should be an option to remove the typedefs or remove the leading`_` from the structs so you can use `struct name` instead of `Name` as required by many codestyles.

PetteriAimonen, 2022-07-04:
This feature is now implemented thanks to @td-krzysiek!

I invite everyone to review the styling implemented.
If there are any changes that you think should be made, I prefer to do them before next stable release.