nanopb/nanopb #206: Add opt-in option to disable bool has_XXX when "optional with default" or required

maruel, 2016-06-10:
Simplify the generated struct by removing the `bool has_FOO` when it is not needed.

**Goals**
- reduce the size of the struct
- improve the alignment, as the embedded bools make everything unaligned;

There's two way to do it; one is to make it as an option at protoc invocation, the other is to make a new extension in NanoPBOptions in nanopb.proto. I don't think adding a new protoc flag to pass via `--nanopb_out=` so I won't describe this option.

**Sample**
```
message foo {
  optional uint32 bar = 1 (nanopb).no_has = true];
  optional uint32 baz = 2 [(nanopb).int_size = IS_16, (nanopb).no_has = true, default = 30];
  optional string fizz = 3 [(nanopb).max_size = 16, (nanopb).no_has = true, default = "yo"];
}
```

**Action items**
- Modify `nanopb.proto`:
  - Add `optional bool no_has = 12 [default = false]` to `NanoPBOptions`.
- Modify `nanopb_generator.py`:
  - Modify `Field.__str__` according to the new rules.
  - Modify `Field.get_initializer` so `FOO_init_default` is generated accordingly.
  - Modify `Field.encoded_size` accordingly.
- Modify `pb_decode.c`:
  - `decode_static_field` in case of `PB_HTYPE_OPTIONAL`?
  - other things I'm not sure.
- Modfiy `pb_encode.c`; not sure what is needed, probably a fair amount of work.

Overall it looks like significant work but on very tight systems it would probably be a significant savings.

PetteriAimonen, 2016-06-10:
This is basically the same underlying work that is needed for issue #182, just different way to set the option. Definitely should do this at some point. The rules for proto3 vs. proto2 are a bit different in that for proto3 one will not encode the field if it equals the hard-coded default value of 0.

Most of the changes would be on the generator side. On the C side one could use `size_offset == 0` to detect whether the field has a has_ field.

maruel, 2016-06-10:
Ok, the main difference with #182 is that this issue is specifically about proto2. The reason I filed the issue is to know if you would be open about this feature? I'm not volunteering right away but may do eventually.

PetteriAimonen, 2016-06-10:
Yes, it seems reasonable.

PetteriAimonen, 2016-09-21:
Related: pull request #215

PetteriAimonen, 2016-10-10:
This is now implemented by #216.

PetteriAimonen, 2016-10-30:
Fix released in nanopb-0.3.7

closed 2016-10-30 by PetteriAimonen