nanopb/nanopb #207: Conflicting include guards for identically-named files from different packages

Ulenspiegel, 2016-07-23:
Given set of proto files in different packages:

```
.
├── annotations
│   └── get
│       └── response.proto    // package annotations.get;
└── info
    └── get
        └── response.proto    // package info.get;
```

nanopb produces two `response.pb.h` files with identical include guard, `PB_RESPONSE_PB_H_INCLUDED`, which prevents using them together:

```c++
#include "annotations/get/response.pb.h"
#include "info/get/response.pb.h"

// ...

annotations_get_Response annotations_response;
info_get_Response info_response;  // Error: info_get_Response not defined.
```

Possible solution I'm now using locally is to add package to the include guard symbol:
```diff
diff -Naur nanopb_generator.py.orig nanopb_generator.py 
--- nanopb_generator.py.orig	2016-07-23 18:31:44.514748236 +0300
+++ nanopb_generator.py	2016-07-23 19:12:51.276700680 +0300
@@ -1045,7 +1045,10 @@
         else:
             yield '/* Generated by %s at %s. */\n\n' % (nanopb_version, time.asctime())
 
-        symbol = make_identifier(headername)
+        if self.fdesc.package:
+            symbol = make_identifier(self.fdesc.package + '_' + headername)
+        else:
+            symbol = make_identifier(headername)
         yield '#ifndef PB_%s_INCLUDED\n' % symbol
         yield '#define PB_%s_INCLUDED\n' % symbol
         try:
```

It'll give guards `PB_ANNOTATIONS_GET_RESPONSE_PB_H_INCLUDED` and `PB_INFO_GET_RESPONSE_PB_H_INCLUDED`. Message IDs (when `PB_MSGID` is defined) are also likely affected.

PetteriAimonen, 2016-07-23:
Including package identifier in the guard name seems like a reasonable fix.

Have to consider what to do about MSGID's. @mhaberler , what do you think?

PetteriAimonen, 2016-10-30:
Fix released in nanopb-0.3.7

closed 2016-10-30 by PetteriAimonen