nanopb/nanopb #210: Callback fields inside of oneof are not supported

rodriguise, 2016-08-02:
I have a very simple message

```
message Variant {
   oneof value {
    string stringValue = 1;
    int32 intValue = 2;
    uint32 uintValue = 3;
    int64 int64Value = 4;
    uint64 uint64Value = 5;
    sint32 sintValue = 6;
    sint64 sint64Value = 7;
    float floatValue = 8;
    bool boolValue = 9;
    bytes byteValue = 10;
    double doubleValue = 11;
  }
}
```

This compiles fine for other langs (google cpp, golang, etc). But I get the following exception with nano.

```
Traceback (most recent call last):
  File "/usr/local/Cellar/nanopb-generator/0.3.5/libexec/nanopb_generator.py", line 1553, in <module>
    main_plugin()
  File "/usr/local/Cellar/nanopb-generator/0.3.5/libexec/nanopb_generator.py", line 1533, in main_plugin
    other_files[fdesc.name] = parse_file(fdesc.name, fdesc, options)
  File "/usr/local/Cellar/nanopb-generator/0.3.5/libexec/nanopb_generator.py", line 1420, in parse_file
    f = ProtoFile(fdesc, file_options)
  File "/usr/local/Cellar/nanopb-generator/0.3.5/libexec/nanopb_generator.py", line 973, in __init__
    self.parse()
  File "/usr/local/Cellar/nanopb-generator/0.3.5/libexec/nanopb_generator.py", line 999, in parse
    self.messages.append(Message(names, message, message_options))
  File "/usr/local/Cellar/nanopb-generator/0.3.5/libexec/nanopb_generator.py", line 785, in __init__
    self.oneofs[f.oneof_index].add_field(field)
  File "/usr/local/Cellar/nanopb-generator/0.3.5/libexec/nanopb_generator.py", line 674, in add_field
    + " (field %s)" % field.name)
Exception: Callback fields inside of oneof are not supported (field stringValue)
--nanopb_out: protoc-gen-nanopb: Plugin failed with status code 1.
```

rodriguise, 2016-08-02:
it seems to not like strings.

PetteriAimonen, 2016-08-03:
You have to specify a maximum size for the string, so that it doesn't need a callback field. See issue #175 for further discussion.

PetteriAimonen, 2016-08-11:
Now I noticed you had tried to use an .options file:

Variant.value.stringValue       max_size:128

But the name will be simply:

Variant.stringValue       max_size:128

closed 2016-08-11 by PetteriAimonen