nanopb/nanopb #212: How to distinguish stream decoding error from insufficient number of bytes?

mxk, 2016-08-08:
I have a byte stream that is made up of length-delimited messages. Nanopb is compiled with PB_BUFFER_ONLY, so I can't use callbacks (though I'm not sure that it would help anyway). I receive N bytes at a time, which may contain zero or more full messages and/or a partial message. Ideally, I want to call `pb_istream_from_buffer`, followed by `pb_decode_delimited` in a loop until the buffer is empty, an error is encountered, or the remaining bytes contain only a partial message, which I would then keep until the next packet arrives.

The problem is that I don't see any way of detecting the last condition and not treating it as an error. I see that `pb_make_string_substream` detects "parent stream too short" condition, and I could copy that same logic calling `pb_decode_varint` myself, but what if it's the delimiter that got split up? In other words, `pb_decode_varint` would also need to distinguish between an invalid (e.g. overflow) and an incomplete varint. Am I missing some obvious way of dealing with this problem? I could count the number of bytes or do the decoding myself, but that will get a bit messy. Even `pb_decode_varint` isn't ideal, because delimiters are 32 bits max, but pb_decode_varint32 isn't exported, so I'd need to do that check myself as well.

PetteriAimonen, 2016-08-09:
I haven't checked in detail, but perhaps checking for
stream->bytes_left == 0 could be made to work.

Currently it doesn't, but what if you do the following changes (which 
could then be merged to nanopb main code):

- Set bytes_left to 0 in pb_read() when "bytes_left < count" is detected.

- Set bytes_left to 0 in pb_make_string_substream() when "bytes_left < size" is detected.

I think that should handle most cases. Varints are read byte-by-byte, so 
when the end of stream is reached bytes_left should already be 0.

This does leave open the possibility of a decoding error on the very 
last byte of the stream being detected as end-of-stream instead. Not 
sure if there is any neat way of fixing that.

mxk, 2016-08-09:
I was thinking of using `pb_decode_tag` to get the length delimiter (by recombining wire_type and tag), which would also give me an eof indication, but that too doesn't distinguish between overflow and eof. In the end, the simplest solution was to check for a complete delimiter first and then use the same logic as `pb_decode_delimited`, treating the failure to create a substream as eof and the failure to decode as a fatal error. In my case, messages longer than 4 KB are considered invalid (that check is done elsewhere), so the delimiter is at most 2 bytes long. Here's what I came up with:

```c
static bool decodeMsg(pb_istream_t* s, Msg* m, bool* err) {
	if (s->bytes_left >= 2) {
		if ((*(uint16_t*)s->state & 0x8080) == 0x8080) {
			*err = true;
			return false;  // Invalid delimiter or message too long
		}
	} else if (!s->bytes_left || *(uint8_t*)s->state != 0) {
		return false; // Incomplete delimiter or message
	}
	pb_istream_t sub;
	size_t n = s->bytes_left;
	if (!pb_make_string_substream(s, &sub)) {
		s->bytes_left = n;
		return false;  // Incomplete message
	}
	*m = (Msg)Msg_init_zero;
	*err = !pb_decode_noinit(&sub, Msg_fields, m);
	pb_close_string_substream(s, &sub);
	return !*err;
}
```

PetteriAimonen, 2016-10-10:
I assume this is now ok for you?

closed 2016-10-10 by PetteriAimonen