nanopb/nanopb #214: Fast shallow has_x = false struct initialization

mxk, 2016-08-11:
nanopb generates X_init_zero struct initializers, which behave like (and may be compiled to) `memset(x, 0, sizeof(*x))`. If the message is very large, as can be the case with messages that behave like unions, where all fields are optional and only one or a few are set at any given time, this form of initialization can get expensive.

The alternative is to manually set has_x = false, x_count = 0, funcs.encoder = NULL, etc. But this is unsafe because a change in the message definition can leave some fields uninitialized.

I was thinking that it could be useful to have either a function or a per-message macro that would mark all optional fields in the outermost struct as not present. A macro would be faster, but I don't know if there are some cases where something like `pb_init_zero(const pb_field_t fields[], void *s)` could be useful as well.

If I find some time, I'll see if I can submit a patch for the macro version, but first I wanted to check if this idea makes sense and if there are some corner cases that I need to be aware of.

PetteriAimonen, 2016-08-11:
Hmm, I would expect memset to be relatively fast compared to any decoding/encoding operations that will be performed on the message.

Its true that for large union messages the has_fields are just a small amount of data compared to the submessages. However, if you only initialize the top-level has_fields, then when you start to fill in one of the submessages, there is again a possibility to miss one of the fields. Well, I guess then you would call a macro for the submessage.. In case of large union messages, using callbacks / oneofs might be preferred to save RAM anyway.

Have you done some benchmarking on this, how much speed benefit is there in your usecase?

mxk, 2016-08-12:
I think you're right; using callbacks for large messages is probably a better way to go. My original problem was the initialization of a struct that may or may not have been encoded later on. The not encoded case was more common (i.e. nothing to send to the other side), so there the full initialization was wasted time. I've modified my code to check whether there is anything to send before doing the initialization, and as you say, encoding is much more expensive in that case.

closed 2016-10-10 by PetteriAimonen