nanopb/nanopb #219: HAS as bit unions

Yakkoo, 2016-10-14:
I had a idea about all HAS_ fields are keeped together as a bit union, and the size is multiply of 8 bits increments, and the name is the same of field.
Have you already think about that? I can try to implement but I am not very expert in Python, do you agree with that?

PetteriAimonen, 2016-10-14:
This has been considered in the past, for example here:
https://github.com/nanopb/nanopb/issues/46#issuecomment-97510661

However it is difficult to provide a nice C API for such bitfields.
So far I've considered two options:

1) C standard bitfield syntax `int foo : 1;`. I haven't found a portable way to read out the bits at runtime, because the placement inside struct is implementation dependent. Also I think some compilers do not support bitfields at all.

2) Declaring `uint8_t has_bits[5];` or similar array and macros like `PB_SET_FIELD(msg, field, value)`. This has actually been implemented once [here](https://github.com/nanopb/nanopb/blob/dynamic_alloc_dev/pb.h#L144). But it just feels too ugly to me. Also it would have to be just an optional feature so that existing code does not break.

Because protobuf 3 does not have the concept of has_ fields, this would only apply for protobuf2. And the memory saving is relatively small, as the field usually takes much more space than a simple 8-bit boolean does.

Yakkoo, 2016-10-16:
but the has are not alligned so they include more bytes for every has, unless we use pack() and then change all the generated code, this is my main concern about actual HAS's that are together with fields, so no align optimization for ARM 32 bits for example

Yakkoo, 2016-10-16:
I just noticed that PACK is already include ( so forget about part of my concern :-) )

PetteriAimonen, 2017-03-05:
I'm closing this off as the current implementation seems like the best tradeoff.

closed 2017-03-05 by PetteriAimonen