nanopb/nanopb #221: pb_ostream as batch

W2K, 2016-11-01:
From what I observe the callback for pb_ostream is called for every 1 byte of message (count in callback). This causes a problem  when sending the byte over Arduno WifiClinet (ESP8266). The message gest received not as one integral array, but as a sseries of single bytes, which makes it impossible to be read.
 One solution would be to use pb_ostream_buffer bt this requires me to allocate fixed buffer or deal with dynamic buffer. Is there a way on the level of nanopb or do I have to deal with this on my own ?

PetteriAimonen, 2016-11-01:
There is no nanopb-level solution for this. I suggest you make your pb_ostream_t callback buffer a group of characters before sending them out, and then provide a separate flush function for it.

Something like:

    typedef struct {
        uint8_t buf[16];
        size_t count;
    } wifi_ostream_state_t;

    bool wifi_ostream_flush(pb_ostream_t *stream)
    {
         wifi_ostream_state_t *state = stream->state;
         size_t count = state->count;
         state->count = 0;
         return wifi_write(state->buf, count);
    }

    bool wifi_ostream(pb_ostream_t *stream, const pb_byte_t *buf, size_t count)
    {
         wifi_ostream_state_t *state = stream->state;
         for (size_t i = 0; i < count; i++)
         {
              if (state->count >= sizeof(state->buf))
              {
                  if (!wifi_ostream_flush(stream))
                      return false;
              }
              state->buf[state->count++] = buf[i];
         }
         return true;
    }

W2K, 2016-11-01:
Thank you  for your suggestion. I appreciate this.  I am looking through it and I see 2 problems.  Correct me if I am wrong but isn't this an equivalent of using pb_ostream_buffer with predefined buffer:

`pb_ostream_t send = pb_ostream_from_buffer(_buffer, sizeof(_buffer));
 auto status = pb_encode(&send, BasicFrame_fields, &writeframe);
_client.write((char*)_buffer,send.bytes_written);`

My main doubt about your posted solution is that if the the amount of data exceds buffer size it wil send incomlete message. In such case pb_osteram_buffer would return error.  I will need to think about some kind of dynamic buffer.

PetteriAimonen, 2016-11-02:
Ah, yeah, if you need to always send the whole message as a single buffer, then pb_ostream_from_buffer() is the best you can have.

You'll have to decide on maximum message length, and allocate long enough buffer for that. I wouldn't recommend a dynamic buffer in most embedded situations, as that will just fail in runtime. If you have enough memory for the maximum message length you need, just allocate that much.

closed 2016-11-19 by PetteriAimonen