nanopb/nanopb #222: Can't install using PlatformIO due to missing version field

a2hill, 2016-11-19:
## Issue
When I attempt `platformio lib install Nanopb` I get an AssertionError: 
>assert set(["name", "version"]) <= set(tmpmanifest.keys())
AssertionError

NanoPB's library.json file is missing a 'version' field which is causing it to not be able to be installed by the PlatformIO Library Manager (version 3.18).

PetteriAimonen, 2016-11-19:
Does it work better now?

For some reason http://docs.platformio.org/en/stable/librarymanager/config.html#version says version number is not required if repository is specified.

a2hill, 2016-11-19:
I know I saw that as well and I was very confused. Maybe it's because the
tag names are prefixed with 'nanopb' so that platformio can't infer the
version number?

a2hill, 2016-11-22:
Yes it installs version 0.3.8-dev now. Unfortunately this fix won't work for anyone wanting to use a previous version. For my project I've only had time to test with 0.3.6 so I've gone ahead and forked that tag, patched the version, and listed my own repository as the dependency. I think this would work as a workaround for anyone else in the same situation.

PetteriAimonen, 2016-11-22:
Hmm, I wonder if it would work if one made duplicate tags out of the old versions?

I have now pushed tags "0.3.6" and "0.3.7" if you want to try it out.

a2hill, 2016-11-22:
Thanks for the duplicate tags, I see them and can install Nanopb using a repo url referencing the tag: 
```
platformio lib install https://github.com/nanopb/nanopb.git\#0.3.6
```
Unfortunately I cannot install Nanopb from the PlatformIO registry using a command like 
```
platformio lib install Nanopb@0.3.6  // Note: this command will work with 0.3.8-dev
``` 
I think this might be due to the possibility that the PlatformIO library crawler has not re-indexed the repo since the change. At this point though I am satisfied that I can reference the library and version by url.

PetteriAimonen, 2017-03-05:
Fix released in nanopb-0.3.8.

closed 2017-03-05 by PetteriAimonen