nanopb/nanopb #225: Error when compiling nanopb.proto with Google's protoc C++ generator.

punitsoni, 2016-11-22:
When trying to generate C++ code using google's protoc generator, I am running into following errors in generated code. 

nanopb.pb.cc: In function ‘void protobuf_AddDesc_nanopb_2eproto()’: 
nanopb.pb.cc:92:3: error: ‘protobuf_AddDesc_descriptor_2eproto’ is not a member of ‘google::protobuf’ 
   ::google::protobuf::protobuf_AddDesc_descriptor_2eproto(); 

Is there any limitation / known issue on nanopb that might cause incompatibility with google's C++ protobuf generator? 

Note: this issue only happens when the API proto file uses nanopb specific options.

closed 2016-11-22 by punitsoni

PetteriAimonen, 2016-11-22:
For future readers, see further discussion on forum: https://groups.google.com/forum/?fromgroups#!topic/nanopb/dx37fW8C9s4

punitsoni, 2016-11-22:
029