nanopb/nanopb #228: proto3 flag doesn't work in proto3 syntax files

elusive-code, 2016-12-22:
Not sure if is a bug report or feature request.

I'm using proto3 syntax file, and I would like to have `has_<field>` generated for a couple of fields 
(not all of them, just a couple).

From these 
https://github.com/nanopb/nanopb/blob/master/generator/proto/nanopb.proto#L71
https://github.com/nanopb/nanopb/blob/master/generator/nanopb_generator.py#L288
https://github.com/nanopb/nanopb/blob/master/generator/nanopb_generator.py#L386
I assumed that adding a `proto3:false` to `.options` file will do the trick, but apparently it didn't - not sure if is a bug, or it was designed only for proto2, but it would be nice to have such an option.

PetteriAimonen, 2016-12-22:
The offending line is this, which sets proto3 automatically for files with proto3 syntax.
https://github.com/nanopb/nanopb/blob/master/generator/nanopb_generator.py#L1383
Though I still think it maybe might be able to override it per-field even currently.. but not sure.

But certainly possible to make this work. However, could you elaborate in what situation you would like to use it? In proto3 generally, missing fields are indicated by setting them to 0. So having a separate has_ field seems redundant.

elusive-code, 2016-12-22:
Yeah, generally, I would agree, however quite often 0 is a valid value, and could not be used as "no-value" flag.

I'm making a protocol to control arduino-based system from PC, and some modules have more then one parameter, which I logically group (for the simplicity mostly).

For example, I use this message to control seven-segment indicator:
```
message IndicatorConfig {
    IndicatorMode mode = 1; //enum , that contains values like "off", "manual", "data_from_sensor1", etc.
    uint32 value = 2;
    uint32 brightness = 3;
    bool blinkingDelimeter = 4;
};
```
Ideally, I would like to be able to set all those parameters at once, as well as change single one of them specifically.

So, for example, if I want to make it display data from sensor, I'm sending only `mode`, if I want to manually set it to certain value, I'm sending `mode` and `value`.

However, when I'm reading that message in arduino, I have no way to tell whether `blinkingDelimeter`=`false` means that I should disable blinking or it was omitted and I shouldn't change current behaviour.

It is definetely possible to have a workaround by making separate messages or including a flag, or using oneof or even making it `proto2 `and marking everything `proto3` except these cases.
However all of these workarounds look quite ugly.

elusive-code, 2016-12-22:
Setting `proto3` flag for `proto3` files seems like a reasonable default :)

However I was not able to override it neither in `.options`, nor within `.proto`
I tested with slightly modified `simple` example
```
syntax = "proto3";

import "nanopb.proto";

message SimpleMessage {
    int32 lucky_number = 1;
    string message = 2 [(nanopb).proto3=false];
}
```
```
SimpleMessage.message   max_size:40,proto3:false
```
I'm using nanopb-0.3.7 (2016-10-30). 
I checked `nanopb.proto` - it does contain 
```
  // Proto3 singular field does not generate a "has_" flag
  optional bool proto3 = 12 [default = false];
```

PetteriAimonen, 2016-12-22:
"quite often 0 is a valid value, and could not be used as "no-value" flag."

I agree on that point, but that is in my opinion a design compromise in proto3. Nanopb cannot affect how the other end will handle the 0 value. So I'm afraid that even if nanopb made a distinction between '0' and 'not present' in proto3 mode, any other proto3 implementation would not understand it.

If you need that distinction, it seems better to either stick to proto2, or to use one of the common workarounds that are used in proto3 (such as oneof, or a separate bool field).

elusive-code, 2016-12-22:
The way nanopb generator works should not affect other implementations in any way.
In fact, it used to generate `has_<field>`s for proto3 and worked fine (as far as I'm aware).
That behaviour changed because of the memory usage considerations
#182, #206

Btw, reference implementations generate `has_<field>` equivalents (for proto3 too).

For the mentioned example I could write in Java something like this:
```
Descriptors.FieldDescriptor blinkingDelimeterField = SBProtocol.IndicatorConfig.getDescriptor()
                .findFieldByName("blinkingDelimeter");
SBProtocol.IndicatorConfig cfg;
...
if (cfg.hasField(blinkingDelimeterField)){
...
}
```

While it is perfectly sensible for nanopb to diverge in that regard by default, considering intended use cases, it would be really nice to have an option to turn it back on for the selected fields.

PetteriAimonen, 2016-12-22:
Hmm, so how does that work in Java? Is it only for reception, or does it affect encoding also?

PetteriAimonen, 2016-12-22:
Ah, indeed there was a bug that made overriding impossible even for proto2 files. That is now fixed.

However if you have patience in explaining details about other proto3 implementations to me, that would be most helpful :) It's always better to understand how exactly people are using nanopb, and how other libraries work with it.

elusive-code, 2016-12-22:
> Hmm, so how does that work in Java?

Ahh, jeez, that was a spot on question.
Apparently it just checks whether value is zero/false/whatever

I just wrote a simple test:
  ```
  @Test
    public void test(){
        SBProtocol.IndicatorConfig cfg = SBProtocol.IndicatorConfig.newBuilder()
                .setMode(SBProtocol.IndicatorMode.IND_OFF)
                .setBlinkingDelimeter(false)
                .build();

        Descriptors.FieldDescriptor modeField = SBProtocol.IndicatorConfig.getDescriptor().findFieldByName("mode");
        Descriptors.FieldDescriptor valueField = SBProtocol.IndicatorConfig.getDescriptor().findFieldByName("value");
        Descriptors.FieldDescriptor brightnessField = SBProtocol.IndicatorConfig.getDescriptor().findFieldByName("brightness");
        Descriptors.FieldDescriptor blinkingDelimeterField = SBProtocol.IndicatorConfig.getDescriptor().findFieldByName("blinkingDelimeter");

        System.out.println("has mode: "+cfg.hasField(modeField));
        System.out.println("has value: "+cfg.hasField(valueField));
        System.out.println("has brightness: "+cfg.hasField(brightnessField));
        System.out.println("has blinkingDelimeter: "+cfg.hasField(blinkingDelimeterField));
    }
```

Results were
```
has mode: false
has value: false
has brightness: false
has blinkingDelimeter: false
```

elusive-code, 2016-12-22:
So yeah, sorry I bothered you, apparently it won't help.

PetteriAimonen, 2016-12-22:
Well, thanks for bothering, because at least it brought out an actual issue affecting proto2/proto3 mixing :)

But yeah, like I expected it won't help much when communicating with other proto3 implementations. Fun thing is that it will work in the "expected" way when nanopb is on the receiving side, i.e. "has_" will indicate whether the field was present in the encoded message or not.

elusive-code, 2016-12-22:
Wait, you mean if I will send that message above to nanopb and check 'has_<fields>' it will return true for `mode` and `blinkingDelimeter` (the ones that were explicitly set to zeros)?

Even if nanopb generated using proto3 syntax? Or only when used with proto2 syntax?

elusive-code, 2016-12-22:
Ahhh, I think I got it. Just checked on the `simple.proto` example how it will be serialized in proto2/proto3
```
syntax = "proto2";
message SimpleMessage {
    optional int32 lucky_number = 1;
}
```

In proto2, when lucky_number set to 0 it is encoded () as 
```
02 08 00 
```
whereas in proto3 it's just
```
00
```
So, if nanopb will receive proto2 message, it will be able to fill those `has_<fields>` correctly

PetteriAimonen, 2016-12-22:
Yeah, that's correct.

And now any field with "proto3 = false" option will be serialized as proto2 field, so it will work for that also. Thanks to the similarity of proto2 vs. proto3, it should also be correctly deserialized by any proto3 parsers which will just wonder "why is this dude sending us 0 fields" ;)

elusive-code, 2016-12-23:
I decided to try workaround with 'oneof' for better compatibility, but it looks like I found an issue when encoding/decoding several oneofs: #229

PetteriAimonen, 2017-03-05:
Fix released in nanopb-0.3.8.

closed 2017-03-05 by PetteriAimonen