nanopb/nanopb #229: Issues encoding/decoding messages that have several 'oneof's

elusive-code, 2016-12-23:
It seems that there is an issue when several oneof fields present in single message

When `simple.proto` example modified like this to simulate `has_<field>` in proto3:
```
syntax = "proto3";

message SimpleMessage {
    oneof has_lucky_number {
        uint32 lucky_number = 1;
    }
    oneof has_counter {
        uint32 counter = 2;
    }
}
```
After running encoding in byte buffer and decoding I receive following output:
```
Encoding 
which_has_lucky_number=1
lucky_number=2
which_has_counter=2
counter=3
Buffer contents:
2 8 2 
Bytes left: 9
Decoded
which_has_lucky_number=1
lucky_number=2
which_has_counter=0
counter=0
```
Encoding is done like this:
```
  message.lucky_number = 2;
  message.which_has_lucky_number = SimpleMessage_lucky_number_tag;
  message.counter = 3;
  message.which_has_counter = SimpleMessage_counter_tag;
....
  /* Now we are ready to encode the message! */
  status = pb_encode_delimited(&stream, SimpleMessage_fields, &message);
```

When using external encoder (tried with java protobuf) nanopb decodes such messages adding values together.
For example, with message like
```
message IndicatorConfig {
    oneof has_mode {
        IndicatorMode mode = 1;
    }
    oneof has_value {
        uint32 value = 2;
    }
    oneof has_brightness {
        uint32 brightness = 3;
    }
    oneof has_blinking_delimeter {
        bool blinking_delimeter = 4;
    }
}
```
sent from java with values
```
{
  mode: IND_MANUAL
  value: 1024
  brightness: 2
}
```
decoded like this:
```
 which_has_mode: 3
 which_has_value: 0
 which_has_brightness: 0
 which_has_blinking_delimeter: 0
 Mode: 1026
 value: 0
 brightness: 0
 blinking_delimeter:0
```

PetteriAimonen, 2016-12-23:
Confirmed for both proto2 and proto3

PetteriAimonen, 2016-12-23:
Hmm yeah, seems to be because there is not explicit indication in pb_fields_t whether several oneof fields belong in same or separate unions. Have to think a bit about how to fix this one.

There is a temporary workaround: add a dummy normal field with tag number that is in between the two oneof values.

PetteriAimonen, 2016-12-26:
Seems like there are two reasonable options:

1) Make generator generate some `PB_FIELD_ONEOF_SEPARATOR` when two oneof fields are detected end-to-end. Would need a way to encode the separator to `pb_field_t` in a way that doesn't get mixed with normal fields.

2) Encode in `pb_field_t` somehow when one oneof ends and next begins. Perhaps either `array_size` field or `data_offset` field could be used for this.

PetteriAimonen, 2016-12-31:
Should be fixed now (using approach 2).

PetteriAimonen, 2017-03-05:
Fix released in nanopb-0.3.8.

galbers, 2018-09-04:
Hello, I am still encountering this exact issue while attempting to decode a message.  I have my test configured using the same example message type that the OP used.  I have found that decoding is failing on the second oneof "has_" field and it's data.

I am using nanopb-0.3.9.1 using "syntax = "proto2"; and have  "#define PB_FIELD_16BIT 1," enabled.

I have tested on 0.3.7, 0.3.8 and of course 0.3.9.1, and still see the same issue across all versions.

PetteriAimonen, 2018-09-05:
Hm, just adding PB_FIELD_16BIT doesn't seem to reproduce this anymore for me.

Could you include the .proto file, the .c program and the binary data you are having trouble with?

galbers, 2018-09-05:
[multiple_onof_test.zip](https://github.com/nanopb/nanopb/files/2354025/multiple_onof_test.zip)

I've attached my entire example source including the proto definitions.  It is a very simple program which sets values into a structure, encodes it into a buffer, then decodes the buffer.  It prints the results on each stage for inspection.

If the zip file for what ever reason doesn't work, I can post the text here.
Thanks! 👍

PetteriAimonen, 2018-09-05:
@galbers It seems your which_ values for the oneofs don't match any of the tags inside the oneof, so it acts as if the field is not set at all. After changing the two lines to:

	commandBuffer.which_command = oneof_command_notify_tag;
	commandBuffer.command.notify.characteristic.which_characteristic = chr_CurrentTemperature_tag;

I get the following output which seems correct:

```
Testing encoding of command
 duid = 1,
 ruid = 2,
 command = 105,
 suid = 4,
 which_characteristic = 4,
 temperature = 22.500000

Encoded length = 18

Raw encoded buffer data:
08 01 10 02 CA 06 0B 08 04 12 07 22 05 15 00 00 B4 41 

Testing decoding of command
Decode length = 18
 duid = 1,
 ruid = 2,
 command = 105,
 suid = 4,
 which_characteristic = 4,
 temperature = 22.500000
```

PetteriAimonen, 2018-09-06:
Good to hear :)

closed 2018-09-06 by PetteriAimonen