nanopb/nanopb #232: Add trait types for c++

eric-wieser, 2017-01-04:
For a message like
```protobuf
message MyMessage {
  float number = 1;
  oneof msg {
    Go go = 2;
    Stop stop = 3;
  }
}
```

It would be great if the following extra template constants existed

* `static pb_field_t const* const pb_traits<MyMessage>::fields == MyMessage_fields`
* `static int const pb_traits<MyMessage>::tag<&MyMessage::number> == MyMessage_number_tag` (C++17)
* `static int const pb_traits<MyMessage>::tag<&MyMessage::number, float> == MyMessage_number_tag` (C++14)
* `static size_t const pb_traits<MyMessage>::max_size == MyMessage_size`
* `static MyMessage const pb_traits<MyMessage>::init_zero == MyMessage_init_zero`
* `static MyMessage const pb_traits<MyMessage>::init_default == MyMessage_init_default`

thirtythreeforty, 2017-02-02:
You can kinda-sorta approximate this by having a macro that specializes all those traits you specify for a given type name, then just calling the macro with all your NanoPB message types.

I agree that this would be handy though.  I'd like to do some type-safe boilerplate reduction in C++ using traits such as these, and it would be nice to have all of it generated.

Note that they should probably be `constexpr`.

vlovich, 2019-03-05:
Add a basic kind of thing in #384. Can be used to expand on this request further.

eric-wieser, 2019-03-06:
Can you give an example of the code yours generates here?

PetteriAimonen, 2019-12-20:
vlovich's pull request is now included in 0.4.0.

closed 2019-12-20 by PetteriAimonen