nanopb/nanopb #233: optional bytes has not has field

Salzi, 2017-01-11:
When I have the following message:

```
message SimpleMessage {
    optional bytes block = 1;
}
```
the generated code looks like this:

```
typedef struct _SimpleMessage {
    pb_callback_t block;
/* @@protoc_insertion_point(struct:SimpleMessage) */
} SimpleMessage;
```
I expected a ```has_block``` boolean within the structure. When I define the max_size in the options it will add that field.
Is this the expected behavior for an unlimited optional bytes field or is this a bug?

PetteriAimonen, 2017-01-11:
Yes, this is expected behaviour.

There are three basic allocation modes for fields in nanopb:
1) Statically allocated storage. For this, the compiler must know the maximum size of each field.
2) Callback functions, which can handle any length
3) Dynamically allocated storage, which needs separately enabled malloc support.

The default is to use 1), except when maximum size is no available it will use 2). With callbacks the has_field is not needed because the callback is just not called when the field is missing. With dynamically allocated fields the NULL value works similarly.

Salzi, 2017-01-11:
Thanks @PetteriAimonen for clarify.

closed 2017-01-12 by Salzi