nanopb/nanopb #234: Remove bundled `plugin.proto` and use `google.protobuf.compiler.plugin_pb`

eric-wieser, 2017-01-16:


PetteriAimonen, 2017-01-16:
Hmm yeah, plugin.proto could probably be debundled now. Back in time plugin interface was experimental and many distributions didn't have it in the python-protobuf package yet.

As for descriptor.proto, it's bundled because I haven't found a good way to determine the include path which would be needed for compiling nanopb.proto.

ptdm, 2018-11-20:
I keep getting this error:
make[2]: *** No rule to make target 'generator/proto/plugin.proto', needed by 'plugin_pb2.py'.  Stop.
CMakeFiles/Makefile2:95: recipe for target 'CMakeFiles/generate_plugin_pb2.py.dir/all' failed
make[1]: *** [CMakeFiles/generate_plugin_pb2.py.dir/all] Error 2

PetteriAimonen, 2018-11-20:
For cmake users, rerunning `cmake` after git pull may be needed.

closed 2019-12-20 by PetteriAimonen