nanopb/nanopb #235: grpc support

d21d3q, 2017-01-17:
I am now testing communication between µC and using via simple protobuf messages and for now I can see that it would be nice to have it defined in grpc style. If it will work fine, I will wrap it in some simple library. 
The question is if it should be part of nanopb, or better make separate project which relays on nanopb?

PetteriAimonen, 2017-01-17:
I think it would make sense to have grpc support in nanopb. I assume it can relatively easily made so that it is an optional feature (e.g. in a separate pb_grpc.c file or similar).

d21d3q, 2017-02-13:
[Here](https://github.com/d21d3q/nanopb/tree/dev-grpc) you can see what direction I took in implementing grpc. @PetteriAimonen now I am going to start to work with generator. In [C++ example of grpc](http://www.grpc.io/docs/tutorials/basic/c.html#generating-client-and-server-code) generator is being split into to separate plugins, but in [python example](http://www.grpc.io/docs/tutorials/basic/python.html#generating-client-and-server-code) there is only one plugin. Would be enough if I would add grpc functionality into current generator (and wrap code in `#ifdef PB_ENABLE_GRPC` ... `#endif`), or should I write separate one?

PetteriAimonen, 2017-02-13:
Seems promising, and reasonable amount of code so far that it could eventually be part of nanopb also.  About the generator, plugin mechanisms may be more trouble than they are worth. But the code could probably be in a separate file that would get imported, if that makes it easier to organize (nanopb_generator.py is getting quite long already).

bl0rq, 2019-02-02:
How far did this get?  I have a need for this and I would be happy to help push things forward.

PetteriAimonen, 2019-02-03:
@bl0rq It's now here https://github.com/d21d3q/nanogrpc but I don't know actual status.

PetteriAimonen, 2022-05-30:
There are currently no plans to implement grpc support in nanopb main code, any support would be a separate project.

closed 2022-05-30 by PetteriAimonen