nanopb/nanopb #238: can not generate c Unions

hualongwang, 2017-01-19:
I want to define an message use oneof, but it can't generate c-code with unions

file: newproto.proto

syntax = "proto2";
import 'nanopb.proto';
message DC_MotorCtrl
{
    required int32  iCtrlType        = 1;  // Range : 1 - 11
    required int32  iSubType       = 2;  // 0: request 1: response
    required int32  iSwitchType   = 3;  // 0x01 - 0x08
    required int32  iSeqNumber  = 4;  // sequence number
    oneof Payload  {
        bool  bStatus = 5;   // 0: success , others : fail
        int32 iSteps  = 6;   // Moving steps 
        int32 iTime   = 7;   // Time
        /*
         * First Byte  : 
         * Second Byte : 
         */
        bytes szTimeSpeed = 8 [(nanopb).max_size = 8];
        /*
         * First  Byte : 
         * Second Byte : 
         * Thrid  Byte : 
         * Fourth Byte : 
         */
        bytes szMoveTrack = 9 [(nanopb).max_size = 16];  
    }
}

generate command :
protoc -I nanopb/generator/proto -I ./ -o newproto.pb newproto.proto
python ../generator/nanopb_generator.py newproto.pb

But in newproto.pb.h , the code generated like use an optional:
/* Struct definitions */
typedef PB_BYTES_ARRAY_T(8) DC_MotorCtrl_szTimeSpeed_t;
typedef PB_BYTES_ARRAY_T(16) DC_MotorCtrl_szMoveTrack_t;
typedef struct _DC_MotorCtrl {
    int32_t iCtrlType;
    int32_t iSubType;
    int32_t iSwitchType;
    int32_t iSeqNumber;
    bool has_bStatus;
    bool bStatus;
    bool has_iSteps;
    int32_t iSteps;
    bool has_iTime;
    int32_t iTime;
    bool has_szTimeSpeed;
    DC_MotorCtrl_szTimeSpeed_t szTimeSpeed;
    bool has_szMoveTrack;
    DC_MotorCtrl_szMoveTrack_t szMoveTrack;
/* @@protoc_insertion_point(struct:DC_MotorCtrl) */
} DC_MotorCtrl;

i try use python ../generator/nanopb_generator.py -s no_unions:false newproto.pb
but it does not work.

PetteriAimonen, 2017-01-19:
It seems to generate an union correctly for me.

Please double-check that you are not actually checking some old version of your newproto.pb.h, because it seems like the .proto you have doesn't match at all:

    bytes szTimeSpeed = 8 [(nanopb).max_size = 8];

vs.

    DC_MotorCtrl_szTimeSpeed_t szTimeSpeed;

Which looks like some other version of the .proto that would have a submessage instead of bytes.

hualongwang, 2017-01-19:
Thanks for your answer.

Does it related with version of protoc and nanopb?  
[hualong@localhost src]$ ./protoc --version
libprotoc 3.1.0

nanopb-0.3.7 (2016-10-30)

PetteriAimonen, 2017-01-20:
Can you try this:

1. Copy-paste the .proto code from your issue report into a .proto file with a new name.
2. Compile that file with -v option given to nanopb_generator.py
3. Paste the command output and the resulting .pb.h here, including the comments at the top of the file.

PetteriAimonen, 2017-02-12:
I'm closing this off because of lack of information. If you still have the problem, we can continue the discusion on the support forum.

seanmc602, 2018-02-07:
I am seeing this issue using nanopb-0.3.9 with libprotoc 3.1.0:

protoc -I ~/projects/nanopb-0.3.9/generator/proto -I . -o msg_union.pb msg_union.proto
nanopb_generator.py -v -s no_unions:false msg_union.pb 
Options file not found: msg_union.options
Options for msg_union.pb: no_unions: false

Options for msg_foo: no_unions: false

Options for msg_foo.data: no_unions: false

Options for msg_foo.size: no_unions: false

Options for msg_foo.name: no_unions: false

Options for msg_bar: no_unions: false

Options for msg_bar.name: no_unions: false

Options for msg_bar.value: no_unions: false

Options for msg_bar.precision: no_unions: false

Options for msg_union: no_unions: false

Options for msg_union.type: no_unions: false

Options for msg_union.foo: no_unions: false

Options for msg_union.bar: no_unions: false

Options for msg_union.msg_type: no_unions: false

Writing to msg_union.pb.h and msg_union.pb.c

msg_union.proto:
```
syntax = "proto2";

import "nanopb.proto";

message msg_foo {
    required bytes data = 1;
    required uint64 size = 2;
    optional string name = 3;
}

message msg_bar {
    required string name = 1;
    optional int32 value = 2;
    optional float precision = 3;
}

message msg_union {
    enum msg_type {
        UNKNOW = 0;
        FOO = 1;
		BAR = 2;
    }

    required msg_type type = 1;

    oneof msg {
     msg_foo foo = 2;
     msg_bar bar = 3;
  }
}
```

msg_union.pb.h:

```
/* Automatically generated nanopb header */
/* Generated by nanopb-0.3.9 at Wed Feb  7 10:38:25 2018. */

#ifndef PB_MSG_UNION_PB_H_INCLUDED
#define PB_MSG_UNION_PB_H_INCLUDED
#include <pb.h>

/* @@protoc_insertion_point(includes) */
#if PB_PROTO_HEADER_VERSION != 30
#error Regenerate this file with the current version of nanopb generator.
#endif

#ifdef __cplusplus
extern "C" {
#endif

/* Enum definitions */
typedef enum _msg_union_msg_type {
    msg_union_msg_type_UNKNOW = 0,
    msg_union_msg_type_FOO = 1,
    msg_union_msg_type_BAR = 2
} msg_union_msg_type;
#define _msg_union_msg_type_MIN msg_union_msg_type_UNKNOW
#define _msg_union_msg_type_MAX msg_union_msg_type_BAR
#define _msg_union_msg_type_ARRAYSIZE ((msg_union_msg_type)(msg_union_msg_type_BAR+1))

/* Struct definitions */
typedef struct _msg_bar {
    pb_callback_t name;
    bool has_value;
    int32_t value;
    bool has_precision;
    float precision;
/* @@protoc_insertion_point(struct:msg_bar) */
} msg_bar;

typedef struct _msg_foo {
    pb_callback_t data;
    uint64_t size;
    pb_callback_t name;
/* @@protoc_insertion_point(struct:msg_foo) */
} msg_foo;

typedef struct _msg_union {
    msg_union_msg_type type;
    bool has_foo;
    msg_foo foo;
    bool has_bar;
    msg_bar bar;
/* @@protoc_insertion_point(struct:msg_union) */
} msg_union;

/* Default values for struct fields */

/* Initializer values for message structs */
#define msg_foo_init_default                     {{{NULL}, NULL}, 0, {{NULL}, NULL}}
#define msg_bar_init_default                     {{{NULL}, NULL}, false, 0, false, 0}
#define msg_union_init_default                   {(msg_union_msg_type)0, false, msg_foo_init_default, false, msg_bar_init_default}
#define msg_foo_init_zero                        {{{NULL}, NULL}, 0, {{NULL}, NULL}}
#define msg_bar_init_zero                        {{{NULL}, NULL}, false, 0, false, 0}
#define msg_union_init_zero                      {(msg_union_msg_type)0, false, msg_foo_init_zero, false, msg_bar_init_zero}

/* Field tags (for use in manual encoding/decoding) */
#define msg_bar_name_tag                         1
#define msg_bar_value_tag                        2
#define msg_bar_precision_tag                    3
#define msg_foo_data_tag                         1
#define msg_foo_size_tag                         2
#define msg_foo_name_tag                         3
#define msg_union_type_tag                       1
#define msg_union_foo_tag                        2
#define msg_union_bar_tag                        3

/* Struct field encoding specification for nanopb */
extern const pb_field_t msg_foo_fields[4];
extern const pb_field_t msg_bar_fields[4];
extern const pb_field_t msg_union_fields[4];

/* Maximum encoded size of messages (where known) */
/* msg_foo_size depends on runtime parameters */
/* msg_bar_size depends on runtime parameters */
#define msg_union_size                           (14 + msg_foo_size + msg_bar_size)

/* Message IDs (where set with "msgid" option) */
#ifdef PB_MSGID

#define MSG_UNION_MESSAGES \

#endif

#ifdef __cplusplus
} /* extern "C" */
#endif
/* @@protoc_insertion_point(eof) */

#endif
```

PetteriAimonen, 2018-02-07:
Hmm, very strange, with your files and nanopb-0.3.9 and protoc 3.1.0 I get:

<pre>
typedef struct _msg_union {
    msg_union_msg_type type;
    pb_size_t which_msg;
    union {
        msg_foo foo;
        msg_bar bar;
    } msg;
/* @@protoc_insertion_point(struct:msg_union) */
} msg_union;
</pre>

Can you post also the intermediate msg_union.pb so I can check whether the difference is on nanopb generator side or on protoc side?

seanmc602, 2018-02-07:
Sure thing, here it is.

[msg_union.pb.gz](https://github.com/nanopb/nanopb/files/1704636/msg_union.pb.gz)

PetteriAimonen, 2018-02-07:
Ok, that file is byte-per-byte identical to what I get, so the difference is in nanopb_generator.py somehow.

Which python version are you using?
I have:

<pre>
petteri@oddish:~/nanopb-test/nanopb-0.3.9-linux-x86$ python
Python 2.7.12 (default, Dec  4 2017, 14:50:18) 
[GCC 5.4.0 20160609] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import google.protobuf
>>> google.protobuf.__file__
'/usr/local/lib/python2.7/dist-packages/protobuf-3.0.0a4-py2.7.egg/google/protobuf/__init__.pyc'
</pre>

and also I get same results with the python bundled in the binary package, which is 2.7 with protobuf-3.1.0.

seanmc602, 2018-02-07:
Samish, 2.7.6
```
scardello@scardello-7510:~/projects/nanopb-0.3.9/generator$ python
Python 2.7.6 (default, Nov 23 2017, 15:49:48) 
[GCC 4.8.4] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import google.protobuf
>>> google.protobuf.__file__
'/usr/lib/python2.7/dist-packages/google/protobuf/__init__.pyc'
>>> 
```

PetteriAimonen, 2018-02-08:
Hmm, reviewing the code, one possible reason would be if your Python protobuf module was older than 2.6.0. But that version is from 2014 so its somewhat unlikely to find it nowadays.

Check first `pydoc google.protobuf` and scroll to the bottom to see the version.

If the version is newer than 2.6.0, try replacing your nanopb_generator.py with the attached one that has a few more debug prints.

[nanopb_generator.txt](https://github.com/nanopb/nanopb/files/1708461/nanopb_generator.txt)

seanmc602, 2018-02-08:
Aha! That was it, my protobuf module was sitting at 2.5.0. I updated to 3.5.1 and now nanopb_generator.py creates proper unions!

Thanks a lot!

PetteriAimonen, 2018-02-09:
Nice to hear that you got it working :)

I added some of the `--verbose` prints to the generator in git master now, so this issue should be easier to figure out in the future.

PetteriAimonen, 2018-04-14:
Fix released in nanopb-0.3.9.1.

closed 2018-04-14 by PetteriAimonen