nanopb/nanopb #240: Unable to translate pb files on OS X on 0.3.7

matsjj, 2017-02-03:
Going through the examples, I got stuck at
```
python nanopb/generator/nanopb_generator.py message.pb

         ********************************************************************
         *** Failed to import the protocol definitions for generator.     ***
         *** You have to run 'make' in the nanopb/generator/proto folder. ***
         ********************************************************************

Traceback (most recent call last):
  File "nanopb/generator/nanopb_generator.py", line 33, in <module>
    import proto.nanopb_pb2 as nanopb_pb2
  File "[...]/nanopb/generator/proto/nanopb_pb2.py", line 26, in <module>
    dependencies=[google_dot_protobuf_dot_descriptor__pb2.DESCRIPTOR,])
TypeError: __init__() got an unexpected keyword argument 'syntax'
```

Calling make did indeed generate two files, but I'm still stuck with above error. 

It works correctly with 0.3.6, even though it's lacking the support for FT_INLINE that I hoped for. :)

PetteriAimonen, 2017-02-03:
`TypeError: __init__() got an unexpected keyword argument 'syntax'`

This usually means that you have multiple incompatible protobuf installations on your computer.
Can you post the output of the following commands:

    which protoc
    protoc --version
    python -c 'import google.protobuf; print(google.protobuf.__file__)'

matsjj, 2017-02-03:
That was quick. Sure!

```
which protoc
/usr/local/bin/protoc

protoc --version
libprotoc 3.2.0

python -c 'import google.protobuf; print(google.protobuf.__file__)'
/usr/local/lib/python2.7/site-packages/protobuf-2.6.1-py2.7.egg/google/protobuf/__init__.py
```

I see. 2.6.1 doesn't match 3.2. I updated the python library and now it works! 

Maybe it's possible to add this as a test to the generator and spit out a warning. 
Thanks for the help!

PetteriAimonen, 2017-02-03:
Hmm yeah, good point about adding more information to the generator error message.
It doesn't know about the protoc version number, but atleast it could detect this one specific error and tell why it happens.

PetteriAimonen, 2017-03-05:
Fix released in nanopb-0.3.8.

closed 2017-03-05 by PetteriAimonen