nanopb/nanopb #242: bug: encoding extension field when object contains only zeroes

lukaszgemborowski, 2017-02-08:
when developing callbacks for extension fields I have noticed a bug (or at least very unexpected behavior), I think example will be the best to visualize the issue (one of my gtest test cases), this is my 'expected' use case of callbacks:

```cpp
struct context {
        context(bool &value) : value (value) {}

        bool &value;
};

bool nanopb_tests_decode_1(pb_istream_t *stream, pb_extension_t *extension, uint32_t tag, pb_wire_type_t wire_type)
{
        context *ctx = (context *)extension->dest;
        ctx->value = true;
        return true;
}

TEST(nanopb, BUG_encode_decode_extension)
{
        bool callback_called = false;
        context ctx(callback_called);
        uint8_t buffer[256];
        IntegerContainer source_field = { 0 };
        Extendable source = { 0 }, destination = { 0 };

        pb_extension_t source_ext = { 0 };
        source_ext.type = &field_a;
        source_ext.dest = &source_field; // all zeroes - this is important
        source_ext.next = nullptr;
        source_ext.found = false;

        source.extensions = &source_ext;

        auto ostream = pb_ostream_from_buffer(buffer, sizeof(buffer));
        ASSERT_TRUE(pb_encode(&ostream, Extendable_fields, &source));

        pb_field_t field = { 0 };
        pb_extension_type_t type = { 0 };
        pb_extension_t destination_ext = { 0 };

        type.arg = &field;
        type.decode  = &nanopb_tests_decode_1;
        type.encode = nullptr;

        destination_ext.type = &type;
        destination_ext.next = nullptr;
        destination_ext.found = false;
        destination_ext.dest = &ctx;

        // 0 bytes written -> decode callback won't be called for field
        EXPECT_NE(0, ostream.bytes_written);

        destination.extensions = &destination_ext;
        auto istream = pb_istream_from_buffer(buffer, ostream.bytes_written);
        ASSERT_TRUE(pb_decode(&istream, Extendable_fields, &destination));

        EXPECT_TRUE(callback_called);
}
```
result of such test case is:
```
nanopb_basic_tests.cc:118: Failure
Expected: (0) != (ostream.bytes_written), actual: 0 vs 0
nanopb_basic_tests.cc:124: Failure
Value of: callback_called
  Actual: false
Expected: true
```
Description: when source_field is zeroed (it has two int32 fields) then bytes_written on ostream is zero. When such buffer is used in pb_decode then decode callback is never called. Expectation would be that decode callback will be called anyway. I guess this is a bug somewhere in encoding.

My test.proto file is:
```
syntax = "proto2";
import "nanopb.proto";

message IntegerContainer {
        required int32 a = 1;
        optional int32 b = 2;
}

message FloatContainer {
        required float c = 1;
}

message Extendable {
        extensions 10 to 100;
}

extend Extendable {
        optional IntegerContainer field_a = 11;
        optional FloatContainer field_b = 12;
}
```

PetteriAimonen, 2017-02-09:
Hmm yeah, seems that the proto3 checks are mistakenly getting applied for extension fields also in proto 2 mode.

lukaszgemborowski, 2017-02-09:
and honestly I do not see any easy fix for that now. Maybe trying to code some additional information in pb_field_t in extension->type->arg; so you can make a proper decision in encode_basic_field(). But I do not know nanopb implementation in many details.

PetteriAimonen, 2017-02-09:
Seems that it is better to change how proto3 support is implemented, as I now realized that it also has the bug that it is not invoked for pointer/callback fields is proto3 files. But I'll have to think about it a bit.

lukaszgemborowski, 2017-02-09:
Take your time. I won't came up with any solution rather soon as my knowledge of the code base is like 2 hours of reading pb_encode.c. But if you need some help with implementation or testing let me know. I do have some spare time sometimes. ;-)

PetteriAimonen, 2017-02-12:
Should be fixed now; found a pretty simple (if not very elegant) fix.

lukaszgemborowski, 2017-02-12:
Working correctly now. Thanks :-)
```
[ RUN      ] nanopb.BUG_encode_decode_extension
[       OK ] nanopb.BUG_encode_decode_extension (0 ms)
```

PetteriAimonen, 2017-03-05:
Fix released in nanopb-0.3.8.

closed 2017-03-05 by PetteriAimonen