nanopb/nanopb #253: Nanopb Encoding Failure

Shams1363, 2017-03-27:
Hi There,

It looks like nanopb has some problems with repeated fields with size 1. In some rare cases it fails to encode. I have an example attached to this message. I tested it with nanopb 3.3, 3.5 and 3.8. All of them failed the same way.

Best,
[CMakeLists.txt](https://github.com/nanopb/nanopb/files/873492/CMakeLists.txt)

PetteriAimonen, 2017-03-28:
Just to verify, is this the same failure you are observing?

```
Program Started
Size of buff 24 ISSI_MultiTouch_MemoryMap_Entry_size 24
Encoding failed: stream full
```

Shams1363, 2017-03-28:
Yes. That is what I see. It looks like nanopb requires 25 bytes.

PetteriAimonen, 2017-03-28:
Copying the relevant parts from .zip to text for future reference:

.proto:

`message X {
        repeated uint32 x1      = 1;  
        repeated uint32 x2            = 2;  
        repeated uint32 x3          = 3;  
        repeated uint32 x4            = 4; 
}
`

.options:
`
X.x1            max_count:1;
X.x2            max_count:1;
X.x3            max_count:1;
X.x4            max_count:1;
`

Generated size:
`#define X_size                                   24`

Actual, correct size: 4 fields * (1 tag byte + 1 length byte + up to 5 value bytes) = 28 bytes. The generator seems to be calculating this for the non-packed format, which actually takes less space in the 1-item (and only in 1-item) case.

Shams1363, 2017-03-28:
How do I choose packed vs non packed? I didn't change anything.

PetteriAimonen, 2017-03-28:
Ah, sorry for the confusion regarding my notes. I was just writing down for myself what was the root cause, packed vs. non-packed is chosen automatically in nanopb and cannot be changed by user.

It was a bug, should be fixed now :)

Shams1363, 2017-03-28:
Thanks a lot. How do I get the patch from you? My system is working on nanopb 3.3. I would prefer to keep that version and just update few lines of code rather than fully upgrading to a new version.

PetteriAimonen, 2017-03-28:
You can click on the commit link above, or take it here as a raw patch:

https://github.com/nanopb/nanopb/commit/25210ac1419567178502e99f005ddb49c4e63d21.patch

Shams1363, 2017-03-28:
Thank you. I just patched it on my system, and it works fine.

shreyasbharath, 2017-09-14:
I have ran into the same problem. Is there a new release with the fix yet?

PetteriAimonen, 2017-09-14:
No, there is no release with the fix yet. I've been too busy..

PetteriAimonen, 2017-09-23:
Fix released in nanopb-0.3.9.

closed 2017-09-23 by PetteriAimonen