nanopb/nanopb #254: "warning: pointer not aligned at address" and not serializing

4ntoine, 2017-03-29:
I'm getting the warning while building and serializing is not working (length = 0).

```
$make app
cc -Wall -Werror -g -O0 -I/Users/asmirnov/Documents/dev/src/nanopb -omessage message.c                   message.pb.c                /Users/asmirnov/Documents/dev/src/nanopb/pb_encode.c   /Users/asmirnov/Documents/dev/src/nanopb/pb_decode.c   /Users/asmirnov/Documents/dev/src/nanopb/pb_common.c  
ld: warning: pointer not aligned at address 0x1000060AA (_libusb_interface_descriptor_msg_fields + 90 from /var/folders/_l/kl9pdj2x50v7416j8htb6w740000gn/T/message-2cbf72.o)
ld: warning: pointer not aligned at address 0x1000060FE (_libusb_config_descriptor_msg_fields + 62 from /var/folders/_l/kl9pdj2x50v7416j8htb6w740000gn/T/message-2cbf72.o)
ld: warning: pointer not aligned at address 0x100006142 (_avrdude_descriptor_msg_fields + 34 from /var/folders/_l/kl9pdj2x50v7416j8htb6w740000gn/T/message-2cbf72.o)
ld: warning: pointer not aligned at address 0x100006126 (_avrdude_descriptor_msg_fields + 6 from /var/folders/_l/kl9pdj2x50v7416j8htb6w740000gn/T/message-2cbf72.o)
```

During app launch i can see serialized buffer length is 0:

```
$./message
Encoded (0 bytes)
Device bus/address is 0/0!

```

Here is how to build and test:

```
make proto
make app
./message
```

Here is the full project (similar to 'simple' in examples).
[libusb_avrdude.zip](https://github.com/nanopb/nanopb/files/878035/libusb_avrdude.zip)

4ntoine, 2017-03-29:
```
$protoc-c --version
protobuf-c 1.2.1
libprotoc 3.2.0
```

4ntoine, 2017-03-29:
nanopb commit `90c7269b634845a6318a7969a600d800aae70e1d`

4ntoine, 2017-03-29:
updated to `651bdc45f2180b17c132470ff1a3a515dbffaa78` and tested - still having the issue

PetteriAimonen, 2017-03-29:
Hmm, sounds like your compiler generates packed structs but the linker doesn't like it. What C compiler and platform are you using?

You can also try defining PB_NO_PACKED_STRUCTS.

4ntoine, 2017-03-29:
mac osx, clang:

```
$gcc --version
Configured with: --prefix=/Library/Developer/CommandLineTools/usr --with-gxx-include-dir=/usr/include/c++/4.2.1
Apple LLVM version 8.1.0 (clang-802.0.38)
Target: x86_64-apple-darwin16.4.0
Thread model: posix
InstalledDir: /Library/Developer/CommandLineTools/usr/bin
```

4ntoine, 2017-03-29:
```
~/Documents/dev/src/nanopb asmirnov$cat ./pb.h | grep PB_NO_PACKED_STRUCTS
 #define PB_NO_PACKED_STRUCTS 1
#if defined(PB_NO_PACKED_STRUCTS)
~/Documents/dev/src/nanopb asmirnov$make clean
~/Documents/dev/src/nanopb asmirnov$make
[ 16%] Building C object CMakeFiles/protobuf-nanopb.dir/pb_common.c.o
[ 33%] Building C object CMakeFiles/protobuf-nanopb.dir/pb_encode.c.o
[ 50%] Building C object CMakeFiles/protobuf-nanopb.dir/pb_decode.c.o
[ 66%] Linking C static library libprotobuf-nanopb.a
warning: /Library/Developer/CommandLineTools/usr/bin/ranlib: warning for library: libprotobuf-nanopb.a the table of contents is empty (no object file members in the library define global symbols)
[ 66%] Built target protobuf-nanopb
[ 83%] Generating plugin_pb2.py
[ 83%] Built target generate_plugin_pb2.py
[100%] Generating nanopb_pb2.py
[100%] Built target generate_nanopb_pb2.py
~/Documents/dev/src/nanopb asmirnov$cd ./examples/libusb_avrdude/
~/Documents/dev/src/nanopb/examples/libusb_avrdude asmirnov$ls
.             .git          README.txt    message.pb.c  message.proto
..            Makefile      message.c     message.pb.h
~/Documents/dev/src/nanopb/examples/libusb_avrdude asmirnov$rm ./message.pb.*
~/Documents/dev/src/nanopb/examples/libusb_avrdude asmirnov$make proto
protoc --plugin=protoc-gen-nanopb=/Users/asmirnov/Documents/dev/src/nanopb/generator/protoc-gen-nanopb --nanopb_out=. message.proto
~/Documents/dev/src/nanopb/examples/libusb_avrdude asmirnov$make app
cc -Wall -Werror -g -O0 -I/Users/asmirnov/Documents/dev/src/nanopb -omessage message.c                   message.pb.c                /Users/asmirnov/Documents/dev/src/nanopb/pb_encode.c   /Users/asmirnov/Documents/dev/src/nanopb/pb_decode.c   /Users/asmirnov/Documents/dev/src/nanopb/pb_common.c  
~/Documents/dev/src/nanopb/examples/libusb_avrdude asmirnov$./message
Encoded (0 bytes)
Device bus/address is 0/0!
```

4ntoine, 2017-03-29:
and BTW nothing except file generation date is changed in generated files (both .c and .h):

```
~/Documents/dev/src/nanopb/examples/libusb_avrdude asmirnov$git diff message.pb.c 
diff --git a/message.pb.c b/message.pb.c
index d51ea54..79fe2d6 100644
--- a/message.pb.c
+++ b/message.pb.c
@@ -1,5 +1,5 @@
 /* Automatically generated nanopb constant definitions */
-/* Generated by nanopb-0.3.9-dev at Wed Mar 29 11:37:37 2017. */
+/* Generated by nanopb-0.3.9-dev at Wed Mar 29 13:38:51 2017. */
 
 #include "message.pb.h"
 
~/Documents/dev/src/nanopb/examples/libusb_avrdude asmirnov$git diff message.pb.h
diff --git a/message.pb.h b/message.pb.h
index 9b37756..38f6bd2 100644
--- a/message.pb.h
+++ b/message.pb.h
@@ -1,5 +1,5 @@
 /* Automatically generated nanopb header */
-/* Generated by nanopb-0.3.9-dev at Wed Mar 29 11:37:37 2017. */
+/* Generated by nanopb-0.3.9-dev at Wed Mar 29 13:38:51 2017. */
 
 #ifndef PB_MESSAGE_PB_H_INCLUDED
 #define PB_MESSAGE_PB_H_INCLUDED
```

PetteriAimonen, 2017-03-29:
Not sure what those alignment warnings are about, but your actual problem is that the fields are optional and you are not setting the `has_` fields.

Try e.g.
```
        message.has_bus_num = true;
        message.bus_num = 1;
        message.has_device_address = true;
        message.device_address = 2;
```

4ntoine, 2017-03-29:
is it required to set according `has_` field? as in `simple` example `has_` value  is NOT set to `true`:
https://github.com/nanopb/nanopb/blob/master/examples/simple/simple.c#L28

PetteriAimonen, 2017-03-29:
The simple example uses a required field. That's pretty much the difference between required vs. optional, optional fields can be missing from the message so the `has_` field indicates whether it is present.

4ntoine, 2017-03-29:
Yes, i know that but it was not clear that one should set `has_...` to true explicitly. Is that correct?

PetteriAimonen, 2017-03-29:
How else would you expect it to work? In plain C struct there is no way to set it automatically.

4ntoine, 2017-03-29:
It's not too obvious and there is at least one opportunity: if the value is different from initial for the type it can be considered as set, not set otherwise. Not a good decision but anyway.

Thanks for explanation and sharing the code!

closed 2017-03-29 by 4ntoine

girtsf, 2017-03-29:
FWIW, not realizing that I have to set the has_* field tripped me up initially as well. I ended up tracing through the code to realize that I needed to do that. It wouldn't hurt to include an example with an optional field, and maybe also include an example with "proto3" field mode enabled.

jimon, 2017-08-18:
For future reference for whoever finds themselves with this issue - when generating 64 bit executables latest linker included in XCode is complaining about pointer in pb_field not being aligned. This is fixable by enabling `PB_NO_PACKED_STRUCTS`.