nanopb/nanopb #255: Is there way to know if optional field such as submessage was in protobuf blob?

Cloudef, 2017-04-05:
I don't see way in API to determine whether submessage was contained in the original message.
We have some legacy code that does something depending whether optional field was included in the message or not. I see pb_decode being aware of this but not exposing it to public in any way.

If not, what would be the best way to go about this?

PetteriAimonen, 2017-04-05:
Are you using proto2 or proto3? In proto2, there is the has_ field. In proto3 there generally isn't a way - not in nanopb and not in other libraries AFAIK. But disabling proto3 for a single field could be a usable workaround.

Cloudef, 2017-04-05:
proto3, proto3 should still encode this information for message type.

PetteriAimonen, 2017-04-05:
Yeah, though whether the message presence makes sense will depend on the library.

How about the option  [(nanopb).proto3=false]  on the message field? It will give a has_ field and still be fully compatible with proto3 in the encoded format.

Cloudef, 2017-04-05:
`Foo bar = 1 [(nanopb).proto3=false];`
`foobar.proto:95:41: Option "(nanopb)" unknown.`

Do I do this right?
Using the binary release of nanopb 0.3.8

There is quite bit of these fields though.

I would also be open for locally hacking the pb_decode to take in callback which gets called for each decoded submessage.
The reliance on submessage existance is some code I want to get rid of eventually anyways.

PetteriAimonen, 2017-04-05:
Ah, you need `import "nanopb.proto"` also. Like this: https://github.com/nanopb/nanopb/blob/master/tests/options/proto3_options.proto

If you have many fields, it may be easier to put the options in a separate .options file where you can use wildcards. For example, `*.msg_* proto3:false`. (https://jpa.kapsi.fi/nanopb/docs/reference.html#defining-the-options-in-a-options-file)

But yeah, you can of course add custom hacks for this also :) It does not seem like a common enough case to warrant a general API, though.

Cloudef, 2017-04-05:
Ah that wildcard method may be actually good enough. Thanks for your help and thanks for nanopb of course :)

closed 2017-04-05 by Cloudef

cristeab, 2022-09-14:
Hi, in Proto Buffer v3.15.0 fields can be marked as optional. Can this feature be used also in nanopb ?

PetteriAimonen, 2022-09-14:
@cristeab Yes, since nanopb-0.4.4. See https://github.com/nanopb/nanopb/tree/master/tests/proto3_optional

Depending on protoc version it may need `--experimental_allow_proto3_optional`. The nanopb side should work automatically without any special settings.