nanopb/nanopb #259: Decoding FT_POINTER submessage with FT_CALLBACK fields can cause segmentation fault

kazuhiro, 2017-05-25:
**Steps to reproduce the issue**

General recipe:
1. Create a protobuf with submessage decoded as FT_POINTER, which contains a field decoded as FT_CALLBACK.
2. Decode message and call pb_release
3. Decode message again

Real case:
1. Hopefully, the attached code will help to reproduce the issue.

[nanopb_issue.zip](https://github.com/nanopb/nanopb/files/1029159/nanopb_issue.zip)

**What happens?**
The second call to pb_decode cause a segmentation fault. Valgrind reports:
```
==30138== Conditional jump or move depends on uninitialised value(s)
==30138==    at 0x10BE70: decode_callback_field (pb_decode.c:619)
==30138==    by 0x10C099: decode_field (pb_decode.c:679)
==30138==    by 0x10C708: pb_decode_noinit (pb_decode.c:915)
==30138==    by 0x10C873: pb_decode (pb_decode.c:959)
==30138==    by 0x10D758: pb_dec_submessage (pb_decode.c:1349)
==30138==    by 0x10BDE5: decode_pointer_field (pb_decode.c:600)
==30138==    by 0x10C082: decode_field (pb_decode.c:676)
==30138==    by 0x10C708: pb_decode_noinit (pb_decode.c:915)
==30138==    by 0x10C873: pb_decode (pb_decode.c:959)
==30138==    by 0x1090D1: main (main.c:47)
```
Looks like pb_set_message_defaults does not clear callbacks for FT_POINTER allocated structures.

**What should happen?**
Message should be parsed twice.

PetteriAimonen, 2017-06-15:
I don't see a reasonable way to actually make callback fields work inside a FT_POINTER message, because the main program won't be able to set the callbacks. Kind of the same problem as in issue #175.

However the segfault can and will be fixed.

PetteriAimonen, 2017-09-23:
Fix released in nanopb-0.3.9.

closed 2017-09-23 by PetteriAimonen