nanopb/nanopb #260: Introduce fixed_count option for fixed sized arrays

jimon, 2017-05-30:
Currently it's only possible to set `max_count` for `repeated` fields, would be nice to have `fixed_count` for cases when we know exactly how many items we will have. This will allow to save space on `_count` field in the struct.

girtsf, 2017-05-30:
+1. I would find this useful as well.

PetteriAimonen, 2017-06-15:
Yeah, seems reasonable. I'll just copy my notes from the forum here so I'll find them later:

FIXED_LENGTH_BYTES is now generated as its own LTYPE. The obvious way
for FIXED_COUNT_REPEATED would be to have its own HTYPE, but there are
no unused values there. But I think that one could just have it be a
repeated field where pb_field_t.size_offset == 0, and then make the
decoder & encoder take the count from pb_field_t.array_size.

PetteriAimonen, 2018-02-15:
Reopening until fix is released.

jimon, 2018-04-15:
@PetteriAimonen wow, great! thanks a lot!

PetteriAimonen, 2018-04-15:
@jimon Most of the thanks really belongs to @oliverlee :)