nanopb/nanopb #261: Serialization/Deserialization of oneof not working

jrozner, 2017-06-03:
I've put together a basic example of trying to serialize and deserialize data that uses oneof and am running into an issue where the one of subtype is not correctly written out or read in. Included is the .proto file as well as the steps needed to reproduce this. Not sure if I'm doing something wrong but it's more or less based on the test in the tests directly. When attempting to build using the same method as I'm using for this I get the same results with the test but don't get the same results when running through scons.

`
syntax = "proto2";

message Heartbeat {
  required uint32 identifier = 1;
}

message PlayerInfo {
  required uint32 team = 1;
  required string name = 2;
  required uint32 nameLength = 3;
}

message Hit {
  required uint32 shooter = 1;
  required uint32 target = 2;
}

message State {
  required uint32 team = 1;
  required bool disabled = 2;
  required bool flag = 3;
}

message Message {
  required uint32 msgType = 1;

  oneof payload {
    Heartbeat heartbeat = 2;
    PlayerInfo playerInfo = 3;
    Hit hit = 4;
    State state = 5;
  }
}
`

main.c
`
#include <stdio.h>
#include <stdlib.h>
#include <string.h>
#include "pb_common.h"
#include "pb_encode.h"
#include "pb_decode.h"
#include "message.pb.h"

#define BUFFER_SIZE Message_size

int main(int argc, char **argv) {
  FILE *out;
  FILE *in;
  uint8_t buffer[Message_size];
  bool status;

  Message message = Message_init_zero;
  message.msgType = 4;
  message.which_payload = Message_playerInfo_tag;
  strcpy(message.payload.playerInfo.name, "lol");
  message.payload.playerInfo.nameLength = 3;

  pb_ostream_t ostream = pb_ostream_from_buffer(buffer, Message_size);
  status = pb_encode(&ostream, Message_fields, &message);
  if (!status) {
    exit(1);
  }

  if ((out = fopen("data.bin", "wb")) == NULL) {
    printf("failed to open file for writing\n");
    exit(1);
  }

  printf("bytes written: %zu\n", ostream.bytes_written);

  fwrite(buffer, ostream.bytes_written, 1, out);
  fclose(out);

  memset(buffer, 0x0, Message_size);

  if ((in = fopen("data.bin", "rb")) == NULL) {
    printf("failed to open file for reading\n");
    exit(1);
  }

  if (fread(buffer, ostream.bytes_written, 1, in) < 1) {
    printf("failed to read input\n");
    exit(1);
  }

  fclose(in);

  Message imessage;

  memset(&imessage, 0xAA, sizeof(imessage));

  pb_istream_t istream = pb_istream_from_buffer(buffer, Message_size);
  status = pb_decode(&istream, Message_fields, &imessage);
  if (!status) {
    exit(1);
  }

  printf("msg type:%d\n", imessage.msgType);
  printf("which payload: %d\n", imessage.which_payload);
  printf("name: %s\n", imessage.payload.playerInfo.name);
  printf("name length: %x\n", imessage.payload.playerInfo.nameLength);

  return 0;
}

`

`sh
protoc -omessage.pb message.proto
nanopb_generator message.pb
`

All of the header and c files (pb_encode.c/h, pb_decode.c/h, pb.h, and pb_common.c/h are copied into the directory and message.pb.h is modified to switch the angle brackets to double quotes in the include line for pb.h since it's not installed at the system level.

It's compiled using the following line tested on both os x and linux (gcc on linux):
`
clang -o main main.c message.pb.c pb_common.c pb_decode.c pb_encode.c
`

and when running the program this is the output:
`
./main
bytes written: 2
msg type:4
which payload: 0
name:
name length: aaaaaaaa
`

jrozner, 2017-06-04:
Disregard. The version of the generator in Homebrew is fairly old and the code it was generating compiled wasn't happy. I've used an up to date version and everything is fixed.

closed 2017-06-04 by jrozner

PetteriAimonen, 2017-06-04:
Hmm, interesting. Could you tell the version numbers, in case someone else hits the same problem?

jrozner, 2017-06-04:
Sure the version I was generating with was 0.3.6. It may also be related to using the c/h files from master but the generated code from an older version. I didn't have any compilation errors but the code that used oneof just flat out didn't work.

PetteriAimonen, 2017-06-04:
Ah, yeah, it is probably related to this commit:
https://github.com/nanopb/nanopb/commit/48f5dd83530a46dd2423277f4488fd4ce9c93b72
I added a note to the migration document now in case someone else runs into it. Normally I would have made the #ifdef give an error on it but I didn't notice the format change (looks like the backwards compatibility test case doesn't have oneofs in it because oneofs themselves were only added into nanopb a year ago).

As for the alignment warnings, you can define `PB_NO_PACKED_STRUCTS` to get rid of it. The reason for those warnings is that by default nanopb aligns structures for minimum code size, which is somewhat slower and some linkers warn about the slowness.