nanopb/nanopb #262: enum sizes across platforms

rolanddixon, 2017-06-14:
As mentioned in issue #164 I am having trouble encoding a message with an enum on a microcontroller and then decoding the message on a windows computer. This issue has been experienced with nanopb release version 0.3.8 with a gcc compiler for the MCU and both Visual Studio 2015 and MinGW compilers on the windows machine. 

On the MCU, an enum is encoded as 1 byte and on the windows machine, the enum is decoded as 4 bytes. The remaining 3 bytes are taken from what ever comes after the enum in memory. The next type to be decoded is therefore decoded with a start memory location 3 bytes off from where it should be.

I have included some code which should reproduce the error. The encoding should be done on an MCU which is difficult to reproduce code for in a standard way. With this in mind, the example I have provided runs in a windows environment and therefore doesn't not reproduce the error but it is indicative of what I am doing on both ends. The following is a visual studio project that contains all relevant .proto and source code: [Test.zip](https://github.com/nanopb/nanopb/files/1074621/Test.zip). This requires an environment variable called NANOPB_PATH to be set up that points to the root directory of nanopb that contains the .h and .c files needed by nanopb. 

Below are the major files:

Test.proto:
```
syntax = "proto3";
import "nanopb.proto";

enum EDeviceId {ERR = 0; DEV1 = 1; DEV2 = 2;}

message RawData
{
	fixed32 timeStamp = 1;
	EDeviceId deviceId = 2; 
	bytes data = 3 [(nanopb).max_size = 6, (nanopb).fixed_length = true];
}

message CalibratedData
{
	fixed32 timeStamp = 1;
	EDeviceId deviceId = 2; 
	float x = 3;
	float y = 4;
	float z = 5;
}

enum ExtType {DUM = 0; RD = 1; CD = 2;}
message Ext
{
	oneof Msg
	{
		// these numbers must match the type enums above
		RawData raw = 1;
		CalibratedData calib = 2;
	}
}
```

Encode.cpp:
```
#include "pb_encode.h"
#include "Test.pb.h"

#include <stdio.h>
#include <string.h>
#include <stdint.h>

/* Just going to open a file supplied by arvg, encode a simple test
* and write the result to file. This should be written on a
* microcontroller to reproduce the error I am getting. Hopefully,
* if it is my implementation problem, this test should help.
*/

/* buffer for serialised data */
uint8_t buffer[256];

int main(int argc, char* argv[])
{
	if (argc < 2)
	{
		printf("Please supply a file name to encode to as an argument\n");
		return -1;
	}

	FILE* pFile = fopen(argv[1], "wb");
	if (pFile == NULL)
	{
		printf("Could not open file\n");
		return 0;
	}

	/* some random test data generated by device DEV1 */
	RawData raw = { (uint32_t)1234, EDeviceId_DEV1,{ (uint8_t)0, (uint8_t)1, (uint8_t)2, (uint8_t)3, (uint8_t)4, (uint8_t)5 } };

	/* the raw data is passed to either a logging stream or communication stream
	* in the form of a Ext with a oneof for encoding and sending.  */
	Ext ext;
	ext.which_Msg = (pb_size_t)EDeviceId_DEV1;
	memcpy(&ext.Msg.raw, &raw, sizeof(raw));

	pb_ostream_s ostrm = pb_ostream_from_buffer(buffer, sizeof(buffer));
	if (!pb_encode_delimited(&ostrm, Ext_fields, &ext))
	{
		printf("An error occured during serialisation\n");
	}

	fwrite(buffer, 1, ostrm.bytes_written, pFile);

	fclose(pFile);

	return 0;
}
```

Decode.cpp:
```
#include "pb_decode.h"
#include "Test.pb.h"

#include <stdio.h>
#include <string.h>
#include <stdint.h>

/* Just going to open a file supplied by arvg, decode a simple test file
 * The file should be written by a microcontroller to reproduce the error.
 * Once again, if the error is in my implementation, then hopefully this
 * example can catch it.
 */

/* buffer for deserialisation */
uint8_t* pBuf = NULL;

int main(int argc, char* argv[])
{
	if (argc < 2)
	{
		printf("Please supply a file name to encode to as an argument\n");
		return -1;
	}

	FILE* pFile = fopen(argv[1], "rb");
	if (pFile == NULL)
	{
		printf("Could not open file\n");
		return 0;
	}

	/* size of file? */
	fseek(pFile, 0, SEEK_END);
	size_t len = ftell(pFile);
	rewind(pFile);

	/* allocate buffer */
	uint8_t* pBuf = new uint8_t[len];
	fread(pBuf, 1, len, pFile);
	fclose(pFile);

	pb_istream_t istrm = pb_istream_from_buffer(pBuf, len);

	Ext ext;
	if(!pb_decode_delimited(&istrm, Ext_fields, &ext))
	{
		printf("An error occured during deserialisation\n");
	}

	delete[] pBuf;

	/* lets do a check as we know the actual values in this case */
	if (ext.Msg.raw.deviceId != EDeviceId::EDeviceId_DEV1)
	{
		printf("This would happen in the real test case as the deviceId is now 4 bytes, whereas it was encoded as 1 byte on the MCU. These four bytes include the top three bytes from the data array.\n");
	}
	for (size_t i = 0; i < sizeof(ext.Msg.raw.data); i++)
	{
		if (ext.Msg.raw.data[i] != i)
		{
			printf("This would happen in the real test as the deviceId enum has taken 3 of the 6 bytes from the array.\n");
		}
	}

	return 0;
}
```

PetteriAimonen, 2017-06-14:
Hmm, did I understand correctly: if you encode the data on the MCU, you get the problem, but if you encode on Windows, you do not? Can you take the binary data given by both so that I could see what difference there is?

Other than that, the symptoms sound very much like your compiler behaves differently for the Test.pb.c file and your Decode.cpp. This sometimes happens if your C++ and C compilers have different configurations. An easy way to test is to rename the .pb.c to .pb.cpp (causing it to be compiled as C++) and see if it has the same problem.

rolanddixon, 2017-06-14:
Yes, you understood correctly. I will send you the binary data tomorrow and have a try with the c++ compiler.

rolanddixon, 2017-06-15:
Hi Petteri

I can't seem to replicate the issue today... This leads me to think that I have a bug in my code. Anyway, I will continue to investigate and if it pops up again I will let you know.

jnread, 2017-06-15:
I think we may be falling over a similar issue.  I can encode a PB with an enum value that is > 127 using protoc under Linux but this is failing to decode using nanopb on Cortex-M3 with Keil.

The same PB with an enum value under 127 decodes just fine; unsurprising really as the field data_size on the target is 1.  The enum field in the encoded PB message is encoded as a 2-byte varint, as you would expect.

Should nanopb handle multi-byte enum varints (ie values over 127) correctly?

PetteriAimonen, 2017-06-15:
@jnread Hmm, your issue on the other hand sounds something like #164. Can you check whether your enum gets defined as unsigned "UENUM" or as signed "ENUM" in .pb.c file? What are the minimum and maximum values in your enum? In what way does it fail to decode?

jnread, 2017-06-19:
Hi @PetteriAimonen. The enum field gets defined as "ENUM".

The enum itself defines values from 0-3, 32-35, 65-68, 96-100, 128-129, and 160-162 (ie. is sparse).  In case it's relevant, the enum is defined in a different file that is imported into this one.

By "failure to decode", I see an "integer too large" error in pb_dec_varint().

        clamped = *(int_least8_t*)dest = (int_least8_t)svalue;

Gives svalue = 0x80, clamped = 0xFFFFFFFFFFFFFF80.

I'll do a little more digging...

PetteriAimonen, 2017-06-19:
@jnread Yeah, seems like #164, and it is probably occurring due to the import. The generator can't find the include file, and thus believes the enum to be signed, while it is unsigned in reality. That is part of the same set of issues as #172, #116.

There are a few workarounds that can be used: 1) adding a -1 value to the enum will force the C compiler to make it signed 16 bits  2) giving include path to nanopb separately like this: `--nanopb_out=-Isubmessage:. test.proto`   3) changing build scripts so that protoc is called within the directory where the included .protos are.

dvescovi1, 2017-06-24:
This is similar to an issue I saw. "Some" microcontrollers compilers encode enums (with field counts less than 128 or 256) as a byte (or 8bits) instead of as an int (most cases 32bits) like the "C" standard specifies as an optimization step. Most cases you can turn this optimization "off" in the compiler options and make enums "ints".

winterscar, 2017-10-08:
I don't know if my issue is related to all this, but when I try to encode a value with an enum = 0, the decoder on the other end does not see the enum at all.

PetteriAimonen, 2017-10-08:
@winterscar Are you using proto3? That might cause that, because 0 is the default value for enums in proto3 and thus is not encoded.

NickeZ, 2019-11-20:
I think many embedded targets by default use `-fshort-enums` which will make gcc pick the smallest possible enum size instead of using `int` which is the standard.

AFAIU nanopb will therefore generate code which will compile to incompatible programs if you compile for `x86_64` and `arm-none-eabi`.

rust bindgen actually generates the following C code for a C-compatible rust enum if you explicitly set the representation:

```rust
#[repr(u8)]
enum TEST {
    One,
    Two,
    Three,
};
```

```c
enum TEST {
  One,
  Two,
  Three,
};
typedef uint8_t TEST;
```

I think it would make sense for nanopb to avoid enums and instead define enums as the correct integer type. But maybe I'm missing something.

PetteriAimonen, 2019-11-20:
@NickeZ Hmm, what kind of incompatibility are you thinking about? The structures will be different sizes in memory, but they are never directly shared between platforms. The protobuf serialization format, on the other hand, is independent of the enum size.

The crux of this issue lies more in whether the enum values are signed or unsigned, which is somewhat hard to figure out if the definition is not visible to generator.

NickeZ, 2019-11-20:
Aha, yeah my problem was actually about generating rust bindings. After I found the `-fshort-enums` it worked out. rust bindgen just assumed `u32` size of enums, when they were `u8`.

I guess it could have worked out of the box if the structs generated by nano pb weren't compiler flag dependent.

PetteriAimonen, 2019-11-20:
@NickeZ Yeah, binding-unfriendliness of nanopb comes up every now and then. My opinion is that using bindings for a protobuf library is a bad idea to begin with and will lead to annoying API, and that it is better to use native protobuf implementation on whatever language one uses.

NickeZ, 2019-11-20:
So you think it is better to parse the protobufs twice than to reuse the code?

PetteriAimonen, 2019-11-20:
@NickeZ Ah well, if you have mixed C/Rust environment anyway, yeah, then it can make sense to create bindings.

NickeZ, 2019-11-20:
Yeah we plan to migrate slowly to rust. Eventually we could probably switch out the protobuf parsing. But so far we are happy with nanopb so no rush.

PetteriAimonen, 2019-12-20:
I think I'll close off this issue, even though the underlying complexity still exists.

In most cases the enum sizes between messages in different files should be handled correctly now. For `nanopb_generator.py`, passing all .proto files into it at once makes sure that the dependencies are available.

closed 2019-12-20 by PetteriAimonen