nanopb/nanopb #264: .cpp and namespace option

dvescovi1, 2017-06-24:
Two features I would like to see are:
1.  .cpp file generation 
2. namespace support

I believe both of these should be rather easy to implement and could be implemented as additional "options" in the .OPTION file.

PetteriAimonen, 2017-06-24:
Somewhat related: #46

dvescovi1, 2017-06-24:
not that interested in full .cpp support right now. Just the ability to generate the output file with the .cpp extension and the 
#ifdef __cplusplus extern "C" {
#endif 
.
.
.
#ifdef __cplusplus
}
#endif
 . under our gcc project .c and .cpp actually invoke different compilers ...why I have no clue.

PetteriAimonen, 2017-06-24:
Yeah, it is pretty normal for .c and .cpp to be slightly different compilers and it does sometimes cause problems.

I guess this is a reasonable thing to do and might provide a path towards real c++ support eventually.

obataku-jabil, 2017-06-27:
@dvescovi1 `generate_header` already takes care of the `extern "C"` directive; I think changing the suffix might be as easy as introducing a new option and adjusting the [single relevant line in `process_file`](https://github.com/nanopb/nanopb/blob/master/generator/nanopb_generator.py#L1552):

`sourcename = noext + options.extension + '.c'`

PetteriAimonen, 2017-09-15:
I explored the namespace support part, seems way too complex to implement as an option within the C generator..

But I guess an option for defining the extension should be easy enough.

PetteriAimonen, 2017-09-15:
Now there are generator options -S and -H, use like:

    protoc "--nanopb_out=-S .cxx -H .hxx:output_dir" foo.proto

The namespace support will have to wait for when #46 is implemented, if ever.

PetteriAimonen, 2017-09-23:
-S and -H options support released in nanopb-0.3.9.

closed 2017-09-23 by PetteriAimonen

g-arjones, 2020-04-29:
> I explored the namespace support part, seems way too complex to implement as an option within the C generator..

@PetteriAimonen Could you please elaborate? I am very interested in that and currently considering forking nanopb. A quick look makes me wonder whether isn't this as simple as wrapping the hole .pb.c/.pb.h files in `namespace { ... }`. Potentially getting rid of some macros in the process... Thoughts?

Also, how do you feel about having an option to make structs anonymous just like oneof/union allows?

PetteriAimonen, 2020-04-29:
@g-arjones If I remember correctly, the issue was mostly referencing to the names within the macros. And the macros aren't there just for fun (although they are fun!), some of them are critical for discovering the platform-dependent struct layout at compile time, others are important to retain back- and forward compatibility without regenerating .pb.h files every time.

I think anonymous structs would break the recursion logic. At least unless you do some funny trick like anonymous union with named struct and anonymous struct inside it.