nanopb/nanopb #268: how to deal with import timestamp?

Kulak, 2017-07-12:
In my .proto file I have line:

import "google/protobuf/timestamp.proto";

and then I use:

message Command {
    google.protobuf.Timestamp issuedDT = 1;
}

The generated C code looks good:

#include <pb.h>
#include "google/protobuf/timestamp.pb.h"

but then Compiler rightfully fails to find "google/protobuf/timestamp.pb.h" include file, because it does not exist.

Please, help as I am not sure how to handle this scenario.

Thank you

Kulak, 2017-07-12:
A short solution to my problem is like that.

Locate "google/protobuf/*.proto" that includes timestamp.proto file.  Compiling an entire "google/protobuf" fails with error 

`PATH=/home/sergei/bin:/home/sergei/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/home/sergei/gw/public/bin:/usr/local/maven/apache-maven-3.3.9/bin:/snap/bin:/home/sergei/500C-ISOM-Trunk/itron/gowspace/bin GOPATH=/home/sergei/500C-ISOM-Trunk/itron/gowspace /home/sergei/500C-ISOM-Trunk/tools/nanopb-0.3.8-linux-x86/generator-bin/protoc -I. --nanopb_out=google/protobuf google/protobuf/*.proto
Traceback (most recent call last):
  File "<string>", line 6, in <module>
  File "__main__.py", line 128, in <module>
  File "__main__protoc-gen-nanopb__.py", line 1664, in <module>
  File "__main__protoc-gen-nanopb__.py", line 1644, in main_plugin
  File "__main__protoc-gen-nanopb__.py", line 1519, in parse_file
  File "__main__protoc-gen-nanopb__.py", line 1058, in __init__
  File "__main__protoc-gen-nanopb__.py", line 1084, in parse
  File "__main__protoc-gen-nanopb__.py", line 870, in __init__
  File "__main__protoc-gen-nanopb__.py", line 757, in add_field
Exception: Callback fields inside of oneof are not supported (field string_value)
--nanopb_out: protoc-gen-nanopb: Plugin failed with status code 1.
`

So, I removed all files and kept timestamp.proto.  Rerunning command resulted in generated timestamp.pb.h and timestamp.pb.c files.  I included files in mbed project and it worked.

PetteriAimonen, 2017-09-15:
Yeah, compiling the whole google/protobuf would probably need a few options in .options.

closed 2017-09-15 by PetteriAimonen